pyvisa/pyvisa #196: Installation

andrew57jm, 2015-11-13:
Trying to install PyVISA 1.8 with a working installation of NI Visa 15.0.1 on Windows 10. 
if I  run python -m visa info i get:

    python -m visa info
    Machine Details:
       Platform ID:    Windows-8-6.2.9200
       Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

    Python:
       Implementation: CPython
       Executable:     C:\Python27\python.exe
       Version:        2.7.10
       Compiler:       MSC v.1500 64 bit (AMD64)
       Bits:           64bit
       Build:          May 23 2015 09:44:00 (#default)
       Unicode:        UCS2

    PyVISA Version: 1.8

    Backends:
       ni:
          Version: 1.8 (bundled with PyVISA)
          #1: C:\WINDOWS\system32\visa32.dll:
             found by: auto
             bitness: 64
             Could not get more info:
                VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could     not be located or loaded.
          #2: C:\WINDOWS\system32\visa32.dll:
             found by: auto
             bitness: 64
             Could not get more info:
                VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

Any version incompatibility that could case the library to not be recognized? visa32.dll and visa64.dll are both in my system32 directory??

hgrecco, 2015-11-14:
I have not tried in Windows 10 yet. But there still the possibility that the pyvisa works but this is a bug in `python -m visa info`. Can you check if you can run a simple example?

andrew57jm, 2015-11-14:
Same error message in a simple program when I create a ResourceManager.

Thanks,
Andrew

jamespetitt, 2015-11-19:
Andrew,
I've seen this issue too on a Windows 7 system.  Looks like an issue with determining 32-bit vs 64-bit.  

Is there a way to force 64-bit VISA on install?

hgrecco, 2015-11-20:
You can force it to use any dll of choice:
```python
>>> import visa
>>> rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll:')
```

If this works, we should make a PR to detect a 64 bit windows and use the right dll.

andrew57jm, 2016-01-14:
Still having this issue with Windows 10 even when I'm explicitly specifying the path to visa64 as above. No one at our site is able to install pyvisa with Windows 10. Here's the error message when trying to create the Resource Manager with a path to visa64.dll:

 Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python34\lib\site-packages\pyvisa-1.8-py3.4.egg\pyvisa\highlevel.py", line 1497, in __new__
    obj.session, err = visa_library.open_default_resource_manager()
  File "C:\Python34\lib\site-packages\pyvisa-1.8-py3.4.egg\pyvisa\ctwrapper\functions.py", line 1225, in open_default_resource_manager
    ret = library.viOpenDefaultRM(byref(session))
  File "C:\Python34\lib\site-packages\pyvisa-1.8-py3.4.egg\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

andrew57jm, 2016-01-15:
User error. All is fine if I run Python from an administrator command prompt as opposed to a standard user command prompt.

hgrecco, 2016-01-15:
I am glad that you managed to make it work ... and also that it is not a bug in PyVISA :-D. 

This is one of the the difficulties of developing and testing PyVISA: there a lot of platforms out there, each with their own particular things. One of the largest changes in version 1.5 was to make it work out of the box in all platforms. We succeeded (until now!) and we need to fix the current state. That is way I have just pushed a new version to detect also the 64bit version. Please put here the output `python -m visa info`. I still need to make PyVISA choose the right version if both are available.

Regarding the administrator issue, it would be nice to help future users. Maybe we can add a note in the docs.

meckdahl, 2016-07-26:
Windows 10 is very stiff on permissions, it also requires writing data from the EXE in the User folder rather than the EXE folder (even application configuration files).  Essentially you are best served by creating a folder in the User's  Documents folder with the name of your application or business, and putting all configurations, logs, etc. in there.  If VISA is installed you can use the LabVIEW Data folder for this, too.

mamunur46, 2017-04-26:
hi i'm writing this after 1 year of latest comment of the thread. my problem is very much similar, using pyvisa 1.8 on windows 10 machine along with NIvisa 15.0. My project is to connect with keithley 2400 source meter by GPIB. It works perfect with Labview programming but with python it always give "VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded." again and again. I tried everything discussed earlier and got same result as everyone talking except error remaining part.

Let me List the things people already did.. after 

> python -m visa info

i got the following

[image]

Tried all three possible option as I found online, 

>  rm = visa.ResourceManager()

> rm = visa.ResourceManager(r'C:\Windows\System32\visa32.dll')

>  rm = visa.ResourceManager(r'C:\Windows\System32\visa64.dll')

whole code was as simple as below:

> import visa

> rm = visa.ResourceManager(r'C:\Windows\System32\visa64.dll')

> rm.list_resources()
('ASRL1::INSTR', 'ASRL2::INSTR', 'GPIB0::24::INSTR')

> inst = rm.open_resource('GPIB0::24::INSTR')

> print(inst.query("*IDN?"))

Details error is like below:

Traceback (most recent call last):
  File "C:\Users\user\Desktop\mamun\ttttt.py", line 6, in <module>
    inst = rm.open_resource('GPIB0::24::INSTR')
  File "C:\Users\user\AppData\Local\Programs\Python\Python35\lib\site-packages\pyvisa-1.8-py3.5.egg\pyvisa\highlevel.py", line 1623, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "C:\Users\user\AppData\Local\Programs\Python\Python35\lib\site-packages\pyvisa-1.8-py3.5.egg\pyvisa\highlevel.py", line 1583, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "C:\Users\user\AppData\Local\Programs\Python\Python35\lib\site-packages\pyvisa-1.8-py3.5.egg\pyvisa\ctwrapper\functions.py", line 1381, in parse_resource_extended
    alias_if_exists)
  File "C:\Users\user\AppData\Local\Programs\Python\Python35\lib\site-packages\pyvisa-1.8-py3.5.egg\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

I know i'm making some very basic mistake here, though don't know whats that. And as this is my very first post here there would be much mistakes in writing. please go easy on me and help.

thanks.

dvincentwest, 2017-04-26:
Have you confirmed that the visa dll is in the location you are specifying? I believe it is the normal place, but let's confirm that first.

mamunur46, 2017-04-26:
yes..there are two files..visa32.dll and visa64.dll
I tried both..same result..

dvincentwest, 2017-04-26:
I'll preface this with I have no idea what the problem is but for the sake of curiosity, what happens if on the command prompt you start python in the system 32 folder, import visa, rm = visa.ResourceManager(). Same thing?

mamunur46, 2017-04-26:
its not a problem with `visa.ResourceManager()`

the problem is actually `rm.open_resource()`  which means pyvisa cant access/locate (whatever i dont know) `visa32/64.dll` library.

dvincentwest, 2017-04-26:
Well I wish I knew the answer. Based on the thread above it seems like a permissions issue in Windows. I am running 64 bit python on more than 3 enterprise windows 7 laptops and have not had this issue. A very painful test would be to try installing VISA libraries in a different non system location, like in your user folder and try again. I don't even know if that's possible because I've never tried. 

Other companies like Keysight/Rhode and Schwartz make their own VISA implementations which you could attempt to access with pyvisa. This should obviously not be necessary but since we don't know what the problem is, it is something you could try.

dvincentwest, 2017-04-26:
As another extreme test, you could try running all of this in Fedora in a VM, which should have access to the USB ports if configured correctly. Again obviously not necessary but... depends on how badly you need it and how much time you have

dvincentwest, 2017-04-26:
@mamunur46 , a more rational suggestion... I would ask this question on Stack Overflow.  You'll have a lot of people with a lot of experience thinking about these kinds of problems looking at the question and are more likely to get a helpful answer if the problem has more to do with python/windows/ctypes and windows permissions, etc.

andrew57jm, 2017-04-26:
> untitled

I had the same problem a while ago. I needed to remove everything and 
reinstall while running python as administrator. That fixed it.

MatthieuDartiailh, 2018-02-25:
Closing, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh

misterb76, 2020-06-01:
I'm having the same problem - with windows 10, python 2.7, any ideas? (tried as admin also).

[image]

meckdahl, 2020-06-24:
> I'm having the same problem - with windows 10, python 2.7, any ideas? (tried as admin also).

See my note from 3 yrs ago above.

It is vague in my memory, but I have 20 yrs of LabVIEW app development, so here is my best stab at it:

VISA has a search path that is set in your ini configuration file or it will go with the default one.  So what I believe the error is actually saying is I walked the search path you have specified (or the default) and the file I am looking for is not there.

So you have two possible solutions:
1.  Fix the search path to where the file is at
2.  Put a copy of the file in the default search path

#2 used to be an easy solution by putting any files searched for in the system folder (which was always searched).
Windows 10 will stop your installer from placing system files there now, and it has new locations to put files.

In Windows 10 opinion the User Folder is the best spot for files that must be found by user installed applications (don't shoot me).  So the easiest way to not fight this is to create your install folder as a sub folder in the installation user's folder and put both your application and any app generated files, ini, configuration, etc.  

There are more professional ways to go about this, but this is the easy road to the best of my memory.

See below for likely better answer:  (url has better formatting than here)

=============== from Pyvisa site:
  https://pyvisa.readthedocs.io/en/1.8/configuring.html

In most cases PyVISA will be able to find the location of the shared visa library. If this does not work or you want to use another one, you need to provide the library path to the pyvisa.highlevel.ResourceManager constructor:

>>> rm = ResourceManager('Path to library')

You can make this library the default for all PyVISA applications by using a configuration file called .pyvisarc (mind the leading dot) in your home directory.

Operating System | Location
-- | --
Windows NT | <root>\WINNT\Profiles\<username>
Windows 2000, XP and 2003 | <root>\Documents and Settings\<username>
Windows Vista, 7 or 8 | <root>\Users\<username>
Mac OS X | /Users/<username>
Linux | /home/<username> (depends on the distro)

For example in Windows XP, place it in your user folder “Documents and Settings” folder, e.g. C:\Documents and Settings\smith\.pyvisarc if “smith” is the name of your login account.

This file has the format of an INI file. For example, if the library is at /usr/lib/libvisa.so.7, the file .pyvisarc must contain the following:

[Paths]

VISA library: /usr/lib/libvisa.so.7

Please note that [Paths] is treated case-sensitively.

STC26, 2024-12-18:
Sorry for my English ...

Here is my feedback for this error:
 "Erreur lors de la détection : VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded."

Finally, I found (with the help of ChatGPT):
In VS Code, Go to the command palette with Ctrl + Shift + P.
Search for Python: Select Interpreter.

In my case the selected interpreter didn't have to correct path, so I selected the onther one and now it works well.

See below the pictures:
with teh error:
[image]

with the problem fixed:
[image]