pyvisa/pyvisa #197: ResourceManager() is taking few minutes to create an object.

apiece82, 2015-11-17:
Hi,

Executing these few lines of code:

import visa
rm = visa.ResourceManager()
print(rm)

the last line 'print(rm)' is taking ~4 minutes to display result. Is this normal?
These are python and pyVisa info on my machine:

C:\python_scripts>python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\me\AppData\Local\Programs\Python\Python35\python
.exe
   Version:        3.5.0
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Sep 13 2015 02:27:37 (#v3.5.0:374f501f4567)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243137
         Spec. Version: 4194304
      #2: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243137
         Spec. Version: 4194304

hgrecco, 2015-11-20:
It is not normal. I have never observed this. Is really the `print` command? It just do: `return 'Visa Library at %s' % self.library_path` which is nothing.

apiece82, 2015-11-23:
The 3rd line is just a print command.
The fact is that the 2nd line 'visa.ResourceManager()' which is taking ~4min to complete.

[image]

as can see in screenshot above, after key-in visa.ResourceManager() the command is doing something before the '>>>' of new command to be displayed for next print command.

hgrecco, 2015-12-09:
The second line is finding and opening the library. In my computers it takes seconds. Maybe is an issue with the 64 bit library and will be fixed when we implement #196. Can you try 
```python
>>> import visa
>>> rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll')
```

apiece82, 2015-12-10:
Have tried with these codes:

>>> import visa
>>> rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll')

same case, about ~4 minutes.

hgrecco, 2015-12-10:
It would be very useful if you can open a debugger to see which line is actually taking so long. I just cannot reproduce it in any machine that I have access.

apiece82, 2015-12-10:
I'm not sure how to do this correctly. This is the output I got when turn 'Debug On':

Python 3.5.0 (v3.5.0:374f501f4567, Sep 13 2015, 02:27:37) [MSC v.1900 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
[DEBUG ON]
>>> import visa
[DEBUG ON]
>>> rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll')
Traceback (most recent call last):
  File "<pyshell#1>", line 1, in <module>
    rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll')
  File "C:\Users\abc\AppData\Local\Programs\Python\Python35\lib\site-packages\pyvisa\highlevel.py", line 1488, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Users\abc\AppData\Local\Programs\Python\Python35\lib\site-packages\pyvisa\highlevel.py", line 1460, in open_visa_library
    return cls(argument)
  File "C:\Users\abc\AppData\Local\Programs\Python\Python35\lib\site-packages\pyvisa\highlevel.py", line 103, in __new__
    obj.library_path = library_path
  File "C:\Users\abc\AppData\Local\Programs\Python\Python35\lib\site-packages\pyvisa\highlevel.py", line 103, in __new__
    obj.library_path = library_path
  File "C:\Users\abc\AppData\Local\Programs\Python\Python35\lib\bdb.py", line 48, in trace_dispatch
    return self.dispatch_line(frame)
  File "C:\Users\abc\AppData\Local\Programs\Python\Python35\lib\bdb.py", line 66, in dispatch_line
    self.user_line(frame)
  File "C:\Users\abc\AppData\Local\Programs\Python\Python35\lib\idlelib\Debugger.py", line 20, in user_line
    self.gui.interaction(message, frame)
AttributeError: 'NIVisaLibrary' object has no attribute 'library_path'
[DEBUG ON]
>>> print(rm)
Traceback (most recent call last):
  File "<pyshell#2>", line 1, in <module>
    print(rm)
NameError: name 'rm' is not defined
[DEBUG ON]
>>>

akabcenell, 2016-06-14:
We're running python 2.7 on windows 7. After replacing our motherboard, we ran into the same issue on a machine where everything had previously worked properly. The pyvisa log looks like:

2016-06-14 12:06:14,493 - pyvisa - DEBUG - No user defined library files
2016-06-14 12:06:14,494 - pyvisa - DEBUG - Automatically found library files: ['C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa32.dll']
2016-06-14 12:06:14,497 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2016-06-14 12:06:14,497 - pyvisa - DEBUG - Created library wrapper for C:\Windows\system32\visa32.dll
2016-06-14 12:10:14,605 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x000000000335CAC8>',) -> 0
2016-06-14 12:10:14,605 - pyvisa - DEBUG - Created ResourceManager with session 4097

So it appears to get caught on the creation of viOpenDefaultRM. Has anyone had this issue or found a solution on this?

akabcenell, 2016-06-21:
We reformatted our drive and reinstalled windows, and it was solved.
We also found that it would take four minutes to instantiate the resourcemanager the first time, but it could then be instantiated instantly if done so again within a few minutes. Maybe there's some issue where it can take a long time to find a file, but then it's cached?

MatthieuDartiailh, 2018-02-25:
Closing, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh

FinK-M, 2018-04-25:
I have just run into this issue as well, but seems to be very little information available. Has any explanation been found since?

tobiasgehring, 2018-12-23:
`ret = library.viOpenDefaultRM(byref(session))` in `def open_default_resource_manager(library):` (`functions.py`) takes so long.

MatthieuDartiailh, 2018-12-23:
What VISA implementation are you using ? NI ? Keysight ? 
This happens inside the VISA shared library and I doubt pyvisa can do anything about it.

tobiasgehring, 2018-12-25:
I am using NI. You are right, it does not seem to be related to pyvisa, but to the underlying visa directly.