pyvisa/pyvisa #198: list resources not working for USB Raw devices

totpic, 2015-11-30:
I want to know the identification number of the USB RAW devices I have connected to the computer.
So I write:
[image]
The USB RAW device I have connected is now listed, but NI-MAX can find it.
[image]

Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.10
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          May 23 2015 09:40:32 (#default)
   Unicode:        UCS2

PyVISA Version: 1.7

Backends:
   ni:
      Version: 1.7 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
      #2: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904

hgrecco, 2015-12-09:
`list_resources` follows the same semantics as NI-VISA which is showing only INSTR devices by default. You can list all resources in the following way:
```python
>>> rm.list_resources('?*')
```

closed 2015-12-09 by hgrecco

totpic, 2015-12-09:
Thanks, It is working now.

NPellet, 2017-08-08:
Helped for me to connect to my GPIB-USB-HS USB to GPIB converter.

Thanks !

(But it should be made clearer in the docs, this is an NI product...)