pyvisa/pyvisa #199: pyvisa on the raspberry

alexxxmg, 2015-12-16:
hi, I'm trying to connect my Raspberry Pi 2B to some instruments,
and I'm having issues. 
I dont think this is connected to *VISA installation problem #149* since the errors are different, but if it is I'm sorry!

```
pip install pyvisa
pip install pyvisa-pi
```
then I tested it with
```
import visa
rm=visa.ResourceManager('@py')
print(rm.list_resources())
```
and got:
```
Traceback (most recent call last):
  File "/home/pi/highfreq/pyvisa test.py", line 7, in <module>
    print(rm.list_resources())
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 238, in list_resources
    raise errors.VisaIOError(errors.StatusCode.error_resource_not_found.value)
VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.
```

The system info is this:
```
pi@optberry:~ $ python -m visa info
Machine Details:
   Platform ID:    Linux-4.1.13-v7+-armv7l-with-debian-8.0
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.9
   Compiler:       GCC 4.9.2
   Bits:           32bit
   Build:          Mar  8 2015 00:52:26 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (2.6)
      TCPIP INSTR: Available 
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 
```

can you help me on this?

thank you!

Alessandro

alexxxmg, 2015-12-16:
my bad - I imagine this was due in fact to the lack of additional modules.
I installed pyusb with `pip install --pre pyusb` (since without the --pre switch it doesnt allow it...)

... but still nothing. That's when I realized that my instrument is trying to talk to the RPi with a USB-GPIB adapter, so I guess I'll have to install linux-gpib too.

And that's too bad, because I didnt find any simple way to do it (the simplest seems this https://xdevs.com/guide/ni_gpib_rpi/ , and simple it is not)

anyway, I guess it was all my fault!

MatthieuDartiailh, 2018-02-25:
Closing, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh