pyvisa/pyvisa #204: pyvisa(-py) is stripping received characters

llntc, 2016-02-16:
I'm trying to control my devices over GPIB using pyvisa / pyvisa-py
GPIB-controller is an Agilent 82357a

When I use GPIB-libs directly, everything seems to work fine, however using visa, characters are missing.

> Machine Details:
>    Platform ID:    Linux-3.16.0-38-generic-x86_64-with-LinuxMint-17.2-rafaela
>    Processor:      x86_64
> 
> Python:
>    Implementation: CPython
>    Executable:     /usr/bin/python
>    Version:        2.7.6
>    Compiler:       GCC 4.8.2
>    Bits:           64bit
>    Build:          Jun 22 2015 17:58:13 (#default)
>    Unicode:        UCS4
> 
> PyVISA Version: 1.9.dev0
> 
> Backends:
>    ni:
>       Version: 1.9.dev0 (bundled with PyVISA)
>       Binary library: Not found
>    py:
>       Version: 0.3.dev0
>       ASRL INSTR: Available via PySerial (2.6)
>       TCPIP INSTR: Available 
>       USB RAW:
>          Please install PyUSB to use this resource type.
>          No module named util
>       USB INSTR:
>          Please install PyUSB to use this resource type.
>          No module named util
>       GPIB INSTR: Available via Linux GPIB (4.0.2)
>       TCPIP SOCKET: Available 
> 

Example:

```
import visa
import Gpib

rm = visa.ResourceManager('@py')
instrv1 = rm.open_resource('GPIB0::22::INSTR')
instrv2 = rm.open_resource('GPIB0::16::INSTR')
instrg1 = Gpib.Gpib(0,22)
instrg2 = Gpib.Gpib(0,16)

#Read IDN using GPIB-lib
instrg1.write('*IDN?')
instrg1.read(100)
'Rohde&Schwarz,FSP-30,837866/036,2.80\n'

instrg2.write('*IDN?')
instrg2.read()
'THURLBY-THANDAR,TSX3510P,0,1.20\n'

#Read IDN using Visa
print instrv1.ask('*IDN?')
RheShazFP3,3860628

print instrv2.ask('*IDN?')
TURLYTADRTX50,,.0
```

Any idea what's going wrong?

mbirthelmer, 2016-02-16:
Is there a reason you're using 'ask' rather than 'query' (the former is
deprecated I believe)?

llntc, 2016-02-16:
Not a particular reason, actually I tried both.
Same output for ask/query or separate write/read.

mbirthelmer, 2016-02-16:
The only other difference I can think of is timing, so maybe pay with the
query_delay parameter. Doesn't really explain what you're seeing though.

hgrecco, 2016-02-18:
I guess that you have run the PyVISA alone with the same output, right? (i.e. without instantiating a GPIB object)

llntc, 2016-02-18:
@mbirthelmer 
Played a bit around with this parameter, but it doesn't seem to do anything in this case (as a bit expected)

@hgrecco 
Yes, indeed. The code above was just a mix of different attempts.

It indeed seems some timing problem, as the output is not always the same. For probably the same reason, I'm not always receiving the termination characters (output appears after timeout)
The writing-part seems to work, for instance I can enable/disable the power supply or set the output voltage. But reading values doesn't.

```
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> inst = rm.open_resource('GPIB0::11::INSTR')
>>> inst.query('*IDN?')
u'TRB-HNA,TX50,,.0'
>>> inst.query('*IDN?')
u'TULYTANDARTX50,,.0'
>>> inst.query('*IDN?')
u'TULBY-TADRTX50,,.0'
>>> inst.query('*IDN?')
u'TULYTADRTX50,,.0'
>>> 

>>> inst.query('V?')   (---> the PSU is at 24V)
u'V2.0'
>>> inst.write('V 10.0')
<StatusCode.success: 0>   (----> voltage is now set to 10V)
>>> inst.query('V?')
u'V1.0'  (----> but readout is incorrect)
>>> 

```

hgrecco, 2016-02-18:
It might be a problem with the implementation of the buffer. I will take a look.

MatthieuDartiailh, 2018-02-25:
Closing as I believe this has likely been fixed. Feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh