pyvisa/pyvisa #206: Pyvisa, Pyvisa-py, linux-gpib on the Raspberry Pi B+

Feargus, 2016-03-02:
Okay, bear with me for a second please :-)
I'm trying to get my Raspberry Pi B+ (up to date version of raspbian jessie) to talk to my Keithley 2400 sourcemeter using a NI GPIB-USB-HS adapter
I installed linux-gpib, and it works fine using ` ibtest`, I can talk to the sourcemeter just fine.

I installed `pyvisa` and `pyvisa-py` via `sudo pip install`.

Just for reference, this looks okay to me

```
pi@raspberrypi:~ $ python -m visa info
Machine Details:
   Platform ID:    Linux-4.1.17+-armv6l-with-debian-8.0
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.9
   Compiler:       GCC 4.9.2
   Bits:           32bit
   Build:          Mar  8 2015 00:52:26 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (2.6)
      TCPIP INSTR: Available
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR: Available
      TCPIP SOCKET: Available

```

Then I tried to run the example program, feeding it to python line by line
```
sudo python
>>> import visa
>>> rm = visa.ResourceManager('@py')
```
until here everything works, no errors, but then the errors start
`>>> rm.list_resources()`
gives me [this](http://imgur.com/gqPbEX9)
I also tried
`>>>inst = rm.open_resource('GPIB0::26::INSTR')`
which does work, no errors thrown
`>>> print(inst.query("*IDN?"))`
then takes more than 10 seconds to complete, and the output is the capital letter "K" repeatedly (about 2 or 3 screens full)
while I was trying these commands in different orders, the sourcemeter twice displayed an "ERROR ID CODE +803", which means "not permitted with output off" see [here](http://forum.keithley.com/phpBB3/viewtopic.php?t=70), which makes no sense here, since I am not measuring anything.
also, when entering
`>>> rm.list_resources()`
again, the output is
`(u´GPIB0::26::INSTR´,)`
which somehow looks like a malformed expression (the "u" is there, and the comma "," too, even tough only 1 device is present

In summary:
- linux-gpib works
- pyvisa and pyvisa-py do something, so they are at least not dead
- But something is wrong

Feargus, 2016-03-03:
Today I had another go at it, and I have a little more information.
I'm not certain that it is a pyvisa-(py) problem, but more a problem of termination characters or similar. However, I understand little of these things, so here are the facts for the moment:
`>>> rm.list_resources()`
consistently gives me this
`(u´GPIB0::26::INSTR´,)`
but only when the device is turned on.  (take note of the "u" and the comma)
When it is turned off I get the error shown above (I assume that behaviour is not intended)
` >>>inst = rm.open_resource('GPIB0::26::INSTR')`
works fine (however, I don't understand what the `INSTR` command is actually good for, since some examples don't contain it)
`inst.write("*IDN?")`
works (the Keithley 2400 switches to remote controlled mode), and the python console tells me `statusCode.success:0`
`inst.read()`
takes a long time to complete (~10-20s) and the result is a string of the form `u'KKKKK......KKK'` (note the starting "u", not included in the inverted commas, similar to before)
subsequent commands all fail, with `statusCode.error timeout:-1073807339`

It seems to me the Keithley is sending the first letter of its identity string "K" without pause.
Changing the terminating character
`inst.term_charm="\n"`
did not help as well

Feargus, 2016-03-03:
So after another bout of fiddling, I was able to resolve most of it.
Turns out the Keithley 2400 has an "SCPI" mode and an "488.1" mode, which are slightly different in terms of how commands have to end.
When using the "SCPI" mode everything seems to work fine so far.

That leaves me with 2 questions:
- The letter "u" still appears in the output of `rm.list_resources`. Why?
- When the device is turned off, `rm.list_resources()` still results in an error, instead of an empty list. Is there any way to change this?

mbirthelmer, 2016-03-03:
The u'...' notation indicates that it is a unicode string (see e.g.
https://docs.python.org/2/howto/unicode.html). You only care about what's
between the quotes normally.

MrHelloBye, 2016-07-21:
And the comma with parentheses shows that it is a tuple. I had this same problem.

MatthieuDartiailh, 2018-02-25:
The failure of the list_resources is fixed on master. 
Closing, feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh