pyvisa/pyvisa #207: query_binary_values() error, read_raw() not one cycle, timeout error

m0v0y3, 2016-03-17:
I am trying to read binary data of the VNA ZNB R&S S-parameter measurement. I used this command to read the binary data : 

`>>> values = vna.inst.query_binary_values('CALCULATE:DATA? SDATA', datatype='d', is_big_endian=True)`

Full program :

```
import numpy as np
import visa
import time

class ZNB_20() :
    #-----------open & close------------    
    def open(self, str):
        """ 
            example : str = 'GPIB0::6::INSTR'
        """
        self.rm = visa.ResourceManager()
        self.inst = self.rm.open_resource	(str,read_termination='\n',send_end=True)
        self.inst.write('*CLS;*ESE 1;*SRE 32')

if __name__ == "__main__":
    print('******Start ZNB 20 model test******')

    vna = ZNB_20()
    vna.open("GPIB1::20::INSTR")
    #read test
    print('=====Read test=======')
    print vna.test_idn()
    
    #Setup VNA
    vna.inst.write("FORMAT REAL, 64;:FORMAT:BORDER NORMAL;*OPC")
    vna.inst.write("SYSTEM:COMMUNICATE:GPIB:RTERMINATOR EOI;*OPC")
    vna.inst.timeout = 5000
    vna.inst.discard_null = False
    vna.inst.chunk_size = 20480
    
    vna.inst.write("INITIATE1:IMMEDIATE;*WAI")
    values = vna.inst.query_binary_values('CALCULATE:DATA? SDATA', datatype='d', is_big_endian=True)`

```

The command was good for small data, that is less than 1600 bytes of data. However, when I tried to measure until 6000 bytes of data it returned timeout error. When I checked using NI MAX, the read_raw() command did not read all data in one cycle even if the chunk_size is 20480 bytes (larger than the output data). This make some bytes are missing from the data.  

This is the stream of the read_raw() when I read 6416 bytes of data, beacuse of the a lot of cycle there are some missing bytes.

```
viRead (GPIB1::20::INSTR (0x00000001), "#46416¿îŠNû.Ó±¿ÒQºôòT...", 20480, 322)
viRead (GPIB1::20::INSTR (0x00000001), "Å´x¿í.0cìÍz?Úu^"]ÊÂ¿ì...", 20480, 48)
viRead (GPIB1::20::INSTR (0x00000001), "W‰y¿ëiÓJ>Ö¹?à.`EŸƒ‡¿ê...", 20480, 87)
viRead (GPIB1::20::INSTR (0x00000001), "H½½Ñ?ä”üZâ.à¿çLÓÄL¦u?...", 20480, 75)
viRead (GPIB1::20::INSTR (0x00000001), ".¿äYF<ý“k?èGx*.¡_¿ãu·...", 20480, 373)
viRead (GPIB1::20::INSTR (0x00000001), "Åçpæ?ïeE+hpú?Âç.8<Æw?...", 20480, 1172)
viRead (GPIB1::20::INSTR (0x00000001), "¿ìI.Nv.8?âÊr“·Ú.¿ìÍAé...", 20480, 276)
viRead (GPIB1::20::INSTR (0x00000001), ":»ò'?Ä£eœp}3¿ñ.\.mÑ’?...", 20480, 35)
viRead (GPIB1::20::INSTR (0x00000001), "ÿ?¸.", 20480, 4)
viRead (GPIB1::20::INSTR (0x00000001), "(YöD»¿ñ@í<r.?¬sàî¬™Ó...", 20480, 293)
viRead (GPIB1::20::INSTR (0x00000001), "¿í-•–.ýƒ¿åV¦mã”Ü¿ìKW....", 20480, 68)
```
Can anyone help me how to make the read_raw() read all data in one cycle? 

Thank you for your help..

```
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 30 Stepping 5, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda\python.exe
   Version:        2.7.11
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan 29 2016 14:26:21 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
```

closed 2016-03-17 by m0v0y3

MatthieuDartiailh, 2016-03-17:
Could you explain what solved the issue ? (for future reference)

m0v0y3, 2016-03-17:
The problem is read termination, if we set

`read_termination='\n'`

`read_raw()` command will read binary output of the VNA not the same as the chunk_size (perhaps when it read \n it stopped). So I solved the problem with `read_termination='None'`, however some instrument need `read_termination='\n'` for several command.