pyvisa/pyvisa #211: Replace ni backend & NI-VISA with IVI-VISA

jonnojohnson, 2016-04-22:
The terms "ni", "NI backend", & "NI-VISA" are used in the documentation and code in various places where they really shouldn't as they don't refer to anything associated to National Instruments. Perhaps the term IVI-VISA would be more appropriate since the IVI foundation is the current maintainer of the VISA standard.
For example this is what I see when l look up my python -m visa info:
```
PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1361381330
         Spec. Version: 5243136
```
The backend type is listed as "ni" but it has nothing to do with National Instruments. The VISA vendor is correctly shown as Keysight which follows the IVI-VISA specification. I think using the term "ivi" or "ivi-visa" type for the backend is more accurate. 

There are also lots of similar references in the docs. I'd be happy to help edit those.

Thanks,

Jonno.

MatthieuDartiailh, 2018-02-25:
I agree it would make sense but we would have to keep supporting 'ni' for backward compatibility. @jonnojohnson would you be ready to do part of the work.

MatthieuDartiailh, 2019-08-21:
Closed in favor of #455 to clarify motivations.

closed 2019-08-21 by MatthieuDartiailh