pyvisa/pyvisa #212: "Error: The device has no langid"

patrickmk, 2016-05-16:
I am attempting to connect to a Tektronix DPO 2002B oscilloscope using PyVISA and PyUSB, and when I attempt to run:
```
address = 'USB0::0x0699::0x0399::<serial>::INSTR'

import visa #import pyVISA
rm = visa.ResourceManager('@py')

print rm.list_resources()
#scope = rm.open_resource(address) #open the oscilloscope
#print scope.query('*IDN?')
```

I get the error of: 

```
Traceback (most recent call last):

  File "<ipython-input-3-8d9e627f2f5d>", line 1, in <module>
    runfile('/home/iago/Documents/Code/test_connection_tektronix.py', wdir='/home/iago/Documents/Code')

  File "/home/iago/anaconda2/lib/python2.7/site-packages/spyderlib/widgets/externalshell/sitecustomize.py", line 699, in runfile
    execfile(filename, namespace)

  File "/home/iago/anaconda2/lib/python2.7/site-packages/spyderlib/widgets/externalshell/sitecustomize.py", line 81, in execfile
    builtins.execfile(filename, *where)

  File "/home/iago/Documents/Code/test_connection_tektronix.py", line 20, in <module>
    print rm.list_resources()

  File "/home/iago/anaconda2/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)

  File "/home/iago/anaconda2/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 231, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])

  File "/home/iago/anaconda2/lib/python2.7/site-packages/pyvisa-py/usb.py", line 169, in list_resources
    serial_number=dev.serial_number,

  File "/home/iago/anaconda2/lib/python2.7/site-packages/usb/core.py", line 830, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)

  File "/home/iago/anaconda2/lib/python2.7/site-packages/usb/util.py", line 315, in get_string
    raise ValueError("The device has no langid")

ValueError: The device has no langid
```
 
This same problem arises from attempting to use open_resource(address)
I was able to successfully connect to a LeCroy oscilloscope over TCP/IP, which leads me to believe there is likely an issue with the langid element from PyVISA, since in the file "util.py" in PyVISA, the value of dev.langids is empty (called in line 311). 

Thank you for your help!

gilandose, 2016-06-08:
Also getting this issue

rakansha, 2016-06-08:
Also getting this issue

sabian2008, 2016-06-22:
Also getting this issue

rakansha, 2016-06-29:
When I edited a file, infact created a file "99-garmin.rules" in /etc/udev/rules.d , with the following content:
SUBSYSTEM=="usb", MODE="0666", GROUP="usbusers"
I no longer got the same error

sabian2008, 2016-06-29:
I confirm that rakasha suggestion works. Although I added (don't know if it was needed) usbusers group and my user into it.

MatthieuDartiailh, 2018-02-25:
This should find its way to PyVISA-py documentation.

MatthieuDartiailh, 2020-09-06:
This is now documented in PyVISA-py

closed 2020-09-06 by MatthieuDartiailh