pyvisa/pyvisa #214: py2exe-compiled executable with optimizations fails

compholio, 2016-05-19:
When optimizations are turned on with py2exe then PyVISA fails to import.  This same situation can be tested without using py2exe by turning on optimizations at the python command line:
`python -OO ./script.py`
Should you do this on a script that imports PyVISA then you'll see the error:
> TypeError: unsupported operand type(s) for +=: 'NoneType' and 'str'

This problem results from the docstrings (`__doc__` attribute) not being populated when optimizations are enabled.  There are several ways to solve this problem, but the easiest is likely to check for `__doc__` being None before attempting to add to the documentation.

MatthieuDartiailh, 2018-02-25:
Closing as I don't see any easy way around this, feel free to re-open if there is more to discuss

compholio, 2018-02-25:
How about this?
[0001-Ensure-that-__doc__-exists-before-adding-to-it.patch.txt](https://github.com/pyvisa/pyvisa/files/1756601/0001-Ensure-that-__doc__-exists-before-adding-to-it.patch.txt)

MatthieuDartiailh, 2018-02-25:
Could you make that into a pull request ?

compholio, 2018-02-25:
Done (https://github.com/pyvisa/pyvisa/pull/318)

commit 21d0cd21 (2018-02-26): Merge #318

318: Ensure that __doc__ exists before adding to it. r=MatthieuDartiailh a=compholio

Fixes #214.
PR #318 (2018-02-26): Ensure that __doc__ exists before adding to it.
Fixes #214.
closed 2018-02-26 by bors[bot]