pyvisa/pyvisa #215: VI_ERROR_RSRC_NFOUND while connecting to a LeCroy using VICP

zdomazet, 2016-06-07:
I am trying to connect to a LeCroy Wavesurfer which is connected to my local network. First, I attempted this code:

`import visa
rm = visa.ResourceManager()
scope = rm.open_resource("VICP::192.168.122.53::INSTR")
print(scope.query("*IDN?"))`

This throws out an error: pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

I followed the bug report thread #168, but no luck. :( Likewise, I read through other related bug reports, as well as several StackOverflow threads, but nothing helped. I did manage to get other types of errors, but without understanding the problem, I don't really consider that an improvement. 

I am using Python 3.5.1 with PyVISA 1.8 running on a Win8.1 computer. If needed, I can provide any other relevant information.

Thanks in advance.

jakob400, 2018-02-08:
I'm having the same problem, if anyone has an idea that would helpful!

MatthieuDartiailh, 2018-02-08:
First could you post the output of `python -m visa info`. In addition have you installed the VICP passport for visa (VICP is not supported as part of standard visa) ? If you have access to NIMax, can you find and communicate with your instrument using it ?

jakob400, 2018-02-08:
The only output I get when I run that is ` Abort trap: 6`. How do you install the VICP passport?

MatthieuDartiailh, 2018-02-08:
This output is really weird ! The VICP passport can probably be downloaded from LeCroy website since they created it. Can you test with NI MAX ?

jakob400, 2018-02-08:
Yeah it is! After scouring the internet for similar cases over the past few days, I'm beginning to feel like someone with an incurable disease. 

To be honest, my situation is a bit different from OP but I'm getting the same output as them when I try using pyvisa-py. 

Here are some of my efforts:

`>>> import visa
>>> rm = visa.ResourceManager()
Abort trap: 6`

Me using pyvisa-py:

`>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm
<ResourceManager(<VisaLibrary('unset')>)>
>>> rm.list_resources()
Traceback (most recent call last):
  File "<pyshell#7>", line 1, in <module>
    rm.list_resources()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 238, in list_resources
    raise errors.VisaIOError(errors.StatusCode.error_resource_not_found.value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.`

I'm a bit concerned by the `<ResourceManager(<VisaLibrary('unset')>)>` line, as it seems like it's not finding the right directory for the VISA library, but I'm also not totally sure how to direct it.

MatthieuDartiailh, 2018-02-08:
If visa info does not work, let's do this manually. 
What operating system are you using ? What Python version and Pyvisa version ? Is VISA (either National instrument or Keysight or Tektronik) installed on your system ? And I don't believe that the pyvisa-py backend can really solve your problem here since it does not implement VICP. However opening  raw socket connection to your instrument you may be able to do some tests. If you want to go in that direction, I would suggest to find some documentation about VICP and in particular what port is used for communication.

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh

gtyhughes, 2023-05-12:
I am having the same issue:

`import pyvisa
from pyvisa.resources import MessageBasedResource

rm = pyvisa.ResourceManager()
lecroy = rm.open_resource('VICP::192.168.100.2::INSTR', resource_pyclass=MessageBasedResource)`

Error: pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

I can connect to the device with NI-MAX (with VXI-11LAN/LXI; appears there is no option to connect with VICP protocol). In Tools Menu > NI_VISA > VISA Options > General Settings >Passports, LeCroyVICPPassport64.dll is enabled. 

Firewall is disabled.

I am using Windows 10, Python 3.8.8, PyVisa 1.13.0, NI-VISA 2023 Q2, VICP passport v1.12. Any help would be much appreciated. Cheers!

MatthieuDartiailh, 2023-05-12:
What address do you use in NI-MAX ?

Alternatively pyvisa-py supports vicp when the pyvicp package is installed.

gtyhughes, 2023-05-16:
From 'Add Network Device', if I select 'Manual Entry of Raw Socket' and enter IP and port 1861 and click 'Validate', it says 'Successfully opened a VISA session to "TCPIP0:192.168.100.2::1861::SOCKET". However, if I open VISA Test Panel and try to execute command *IDN?\n as a query, it fails with following (even after increasing timeout to 10 seconds): 

Read Operation
Error
VISA:  (Hex 0xBFFF0015) Timeout expired before operation completed.

I don't think NI MAX is using VICP protocol in this case, though. I don't see a way to force it to use it when adding a network device.

I'll check out pyvisa-py with pyvicp, thanks for the info

gtyhughes, 2023-10-04:
pyvisa-py with pyvicp is too slow for my application. The VICP passport is working when I use the following code (from bug report [# 168](https://github.com/pyvisa/pyvisa/issues/168)):

```
import visa
visa.log_to_screen()
rm = visa.ResourceManager()
sess, ret = rm.open_bare_resource('VICP::169.254.201.2::INSTR')
print(rm.visalib.write(sess, '*IDN?\n'))
print(rm.visalib.read(sess, 128))
```
 But when I try:
    
`lecroy = rm.open_resource('VICP::192.168.100.2::INSTR', resource_pyclass=MessageBasedResource)`

I get the the following:

`VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.`

Any insight would be greatly appreciated. Thanks!

MatthieuDartiailh, 2023-10-18:
Could you post the output of your script with logging enabled ?

I am wondering if the normalization of VICP resource names may be messing things up. You could try to comment the board attribute of VICPInstr in the rname.py and see if it helps.

MatthieuDartiailh, 2023-10-18:
Can you make a PR ? double checking the Lecroy documentation it appears VICP deviates from other resources and does not include a board field.

imirg, 2023-10-19:
> Can you make a PR ? double checking the Lecroy documentation it appears VICP deviates from other resources and does not include a board field.

Yes, I can (try to) do the PR. I will remove the "board" attribute from the VICPInstr completely.

imirg, 2023-10-23:
> Can you make a PR ? double checking the Lecroy documentation it appears VICP deviates from other resources and does not include a board field.

I have tried to implement the changes but I have difficulties passing the tests.

My first solution was to remove the "board" attribute from the VICPInstr class completely. And remove "board" from everywhere in the vicp tests. But I cannot pass the tests with this approach. I get the error saying that there are too many parts for too little attributes after parsing the string with the resource name:
[image]

I believe the error originates from the `_ResourceNameBase` method `from_string`:

```
parts: List[str]
if len(resource_name) == len(interface_type):
   parts = []
else:
   parts = resource_name[len(interface_type) :].split("::")
```
On the output of this code, one obtains a tuple with two elements: an empty string (that corresponds to the absent board number) and a host name. But there is only one attribute left in the VICPInstr class, namely host name. I think the code is not flexible for removing / adding attributes to the resource classes. 

The second solution was to set the "board" in the VICPInstr class to the empty string "" instead of 0. With this I tried to overcome the limitations of removing the attribute from the class. This approach works with the physical hardware, but it does not allow to pass the test either. There is a check for the parameters not to be empty and with an empty string it is not possible to pass through it.
[image]

I do not have a broad overview over the code, so I am not sure how I could fix this easily. Maybe you can see directly some easy solution for this.

MatthieuDartiailh, 2023-10-24:
Actually, because every other resource supports something like a board attribute, the base class make it somewhat mandatory. I tried and pushed a solution to #215. Please have a look and check it fixes your issue.

imirg, 2023-10-24:
> Actually, because every other resource supports something like a board attribute, the base class make it somewhat mandatory. I tried and pushed a solution to #215. Please have a look and check it fixes your issue.

I have checked the functionality on the hardware and it works. I think your fix is doing the job.

PR #780 (2023-10-25): rname: fix VICP resource name normalization
Closes #215