pyvisa/pyvisa #216: write function not taking variables

rakansha, 2016-06-13:
I am using py visa to control wavestation 2052 from my Ubuntu 16.04
my code is:
*import visa
import time
rm=visa.ResourceManager('@py')
lecroy=rm.open_resource('USB0::1535::2593::LCRY2362C02004::0::INSTR')
lecroy.write('C2:OUTP ON')
lecroy.write('C2:OUTP ON,LOAD,HZ')
myfreq=[50,60,70,80,90,100,120,130,140,150]
for number in myfreq:
                     lecroy.write('C2:BSWV TYPE,SINE,FRQ,number,AMP,0.2')
                     time.sleep(20) *
Now it would just change the frequency of channel 2 to 1uHz.
And this happens everytime I send a variable. Say I assigned i=89, and sent i as an argument to lecroy.write('C2:BSWV TYPE,SINE,FRQ,i,AMP,0.2'), the same thing happens
if I send 89 directly or any number directly, it gets reflected and thats not a problem but sending variable is always setting the frequency to 1u and nothing happens after that.
Can anyone help? I need to automate this process of changing frequency to test my amplifiers

geggo, 2016-06-13:
write() expects a string to send to your device. You have to convert the value of the variable to a properly formatted string, try e.g. 
lecroy.write('C2:BSWV TYPE,SINE,FRQ,%d,AMP,0.2'%number)

rakansha, 2016-06-13:
Thanks!!!! It worked

closed 2018-02-25 by MatthieuDartiailh