pyvisa/pyvisa #219: no module named visa

rakansha, 2016-06-26:
even after installing pyvisa, pyusb, python-py
on typing "python -m visa info" in terminal, I get "No Module named visa"
What is the proper way to remove all these and install it again?
I had previously installed it and used it to connect to my instruments,it was all running smoothly until after I changed some file contents ("usb.py") by mistake.Then due to some reasons I tried removing these packages and installing them ..But even after installing them I get "no module named visa"
Please help

mbirthelmer, 2016-06-26:
You can look at the sys.path variable (see
http://stackoverflow.com/questions/135035/python-library-path) to see what
directories Python uses to find your libraries. If your pyvisa stuff isn't
there, then that's the problem.

Depending on how you installed the libraries (a Linux distro package
manager, from source, etc.) things may get dumped into different
directories (especially if you have multiple versions of python in your
system).

MatthieuDartiailh, 2018-02-25:
Closing feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh