pyvisa/pyvisa #221: Value error operation timed out

rakansha, 2016-06-29:
My software details for VISA are:
Machine Details:
   Platform ID:    Linux-4.4.0-28-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.11+
   Compiler:       GCC 5.3.1 20160413
   Bits:           64bit
   Build:          Apr 17 2016 14:00:29 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.dev0
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

I am using py visa to connect to a Keysight 33500B Series waveform generator.I have already successfully used python VISA to connect to Tektronix oscilloscope as well as Teledyne LeCroy Waveform generator..But with Keysight I get an operation time out error on any query or write function..This is the what my input output session looks like:
--------------------------------------------------------------------------------------------------------------------------------
>>> import visa
>>> rm=visa.ResourceManager('@py')
>>> rm.list_resources()
(u'USB0::2391::11271::MY52814581::0::INSTR',)
>>> keysight=rm.open_resource('USB0::2391::11271::MY52814581::0::INSTR')
>>> keysight.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/manish/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/home/manish/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/home/manish/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
>>> 
--------------------------------------------------------------------------------------------------------------------------------

MatthieuDartiailh, 2016-06-29:
Did you properly set the write_termination and read_termination ? To check that the write_termination is correct try sending a command modifying the state of the instrument and check visially that it worked.

rakansha, 2016-06-29:
can you elaborate on how that can be done

rakansha, 2016-06-29:
this is what i get:
----------------------------------------------------------------------------------------------------------------------------------
 print keysight.read_termination
None
----------------------------------------------------------------------------------------------------------------------------------

rakansha, 2016-06-29:
Even write commands are not working.
"Operation timed out " error on those also

MatthieuDartiailh, 2016-06-29:
The manual of the instrument should specify what is the expected termination of a message when writing and reading (line feed \n, carriage return \r, both,  something else...).
Once you know what is/are the terminations character(s), you can set write_termination and read_termination accordingly, this pyvisa will add the write_termination to your messages and strip the read_termination from your messages.

```python
keysight.write_termination = '\r\n'
keysight.read_termination = '\r\n
```

rakansha, 2016-06-29:
I tried all possible combinations : \n,  \r, \r\n, \f
nothing seems to work.

MatthieuDartiailh, 2016-06-29:
Can you communicate using NIMax ? I noticed you are using the py backend which is not yet super stable, so if you can try first to use the ni backend.
Did you find anything in the instrument manual ?

rakansha, 2016-06-29:
actually NI VISA library is not available for my Linux version

calef800, 2017-06-19:
Has this issue been solved. I am trying to connect a raspberry pi to a keysight 33511B function generator. I have tried writing a udev rule (as explained here: https://github.com/pyvisa/pyvisa-py/issues/3) and that didn't work. Seems like there is a common trend with Keysight (Agilent) instruments as it is here as well: https://github.com/pyvisa/pyvisa-py/issues/71. Trying to change the termination character also didn't work for me. You said the teledyne lecroy waveform generator worked quite well?

MatthieuDartiailh, 2017-06-20:
Not to my knowledge but apparrently python-ivi usbtmc works so it might be worth giving it a look and use it to update the current implementation. Do you want to try ? Otherwise I might have some time this week but I cannot make any promise.

calef800, 2017-06-20:
I have tried today and the instrument now receives commands. part of the problem may have been the an incorrect group/user for the udev rules which i made sure to follow the instructions on the usbtmc repo: https://github.com/python-ivi/python-usbtmc However I get errors for some of the scpi commands i send. But that could be due to sending incorrect commands. i am gonna try and check the error messages in the function generator to see whats going on. Thank you for pointing me in the right direction 👍

MatthieuDartiailh, 2017-06-20:
It may be worth adding a note in the docs about the configuration of the udev. At least a note pointing to python-usbtmc. Could you do a PR updating the docs ?

calef800, 2017-06-20:
Yup. I can provide a link to usbtmc and another link to explain creating user groups etc. with a small explanation of what information is where. Would it be best to create a new doc just for explaining udev rules?

calef800, 2017-06-20:
To be clear only the usbtmc library worked. Not pyvisa. I still get a timeout error

MatthieuDartiailh, 2017-06-20:
Interesting in that case it is still worth investigating on the Pyvisa side. I may have some time to give a look tomorrow but in the meantime a PR about udev would still be meaningful so please go ahead with the doc PR.

MatthieuDartiailh, 2017-06-21:
Ideally the PR should be against pyvisa-py. I had no time to actually dive into the differences between pyvisa and python-usbtmc but it looks like python-usbtmc has a number of safeguards against instruments not behaving as expected, so it is probably really worth trying to adapt it here.

ChadCarl30, 2017-07-13:
I am also experiencing this issue with the Tektronix MSO3014 oscilloscope using the pyvisa library. Have you investigated any further on this end?

RobertAdamMcGill, 2019-07-23:
I'm new to this so take my answer with a grain of salt, but I also had a problem with the "Operation timed out" error when using the Tektronix MSO3014 with pyvisa. I fixed it by installing the MSO3000(etc.) IVI driver v1.4 from the Tektronix website:
 
https://www.tek.com/oscilloscope/dpo2002-software 

and following the installation instructions. Hopefully, that will help? If you need more information, please let me know what I should provide - I'm a novice.

mcejp, 2020-02-05:
Same problem with Keysight B2911A SMU on Linux and these versions:
```
Machine Details:
   Platform ID:    Linux-4.15.0-55-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.9
   Compiler:       GCC 8.3.0
   Bits:           64bit
   Build:          Nov  7 2019 10:44:02 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

Any attempt to write to the device would result in `usb.core.USBError: [Errno 110] Operation timed out`

What worked for me: installing the full _Keysight IO Libraries Suite 2020 (for Linux)_ (bit of a hassle, requires a specific kernel version) and explicitly instantiating `visa.ResourceManager("/usr/lib/x86_64-linux-gnu/libivivisa.so")`.

anderson-pa, 2020-03-06:
Having the same issue with an Agilent (Keysight) 33521A but only **after** updating the firmware from v1.14 to v5.02, so that may be a clue. Times out on write and query operations for all combinations of termination characters ('\r', '\r\n', '\n')

My pyvisa info:
> Machine Details:
   Platform ID:    Linux-4.15.0-88-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64
>
> Python:
>    Implementation: CPython
>    Executable:     /usr/bin/python3
>    Version:        3.6.9
>    Compiler:       GCC 8.3.0
>    Bits:           64bit
>    Build:          Nov  7 2019 10:44:02 (#default)
>    Unicode:        UCS4
> 
> PyVISA Version: 1.10.1
> 
> Backends:
>    ni:
>       Version: 1.10.1 (bundled with PyVISA)
>       #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
>          found by: auto
>          bitness: 64
>          Could not get more info:
>             VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
>    py:
>       Version: 0.3.1
>       ASRL INSTR: Available via PySerial (3.4)
>       USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
>       USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
>       TCPIP INSTR: Available 
>       TCPIP SOCKET: Available 
>       GPIB INSTR:
>          Please install linux-gpib to use this resource type.
>          No module named 'Gpib'