pyvisa/pyvisa #223: "list_resources" empty but can still connect explicitly using "open_resource"

christofferTP, 2016-06-30:
Python 2.7.9 (default, Dec 10 2014, 12:24:55) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm = visa.ResourceManager()
>>> rm.list_resources()
()
>>> rm.open_resource('TCPIP0::K-34461A-17982::inst0::INSTR')
<'TCPIPInstrument'(u'TCPIP0::K-34461A-17982::inst0::INSTR')>
>>>

What gives? Can resources be blocked by other applications, not allowing them to be read by list_resources, or is this an issue on my part? I have drivers and I/O libraries installed for the device im using (KeySight 3446X)

MatthieuDartiailh, 2018-02-25:
I have no idea what could have caused that. Will try to look into it.

RognanCr, 2019-04-08:
i have exactly the same problem

MatthieuDartiailh, 2019-04-08:
Could you post the output of `pyvisa-info` ? and the model of the instrument you are trying to connect to ?

RognanCr, 2019-04-09:
Model of instrument is keithley 34461A. If i use `list_resources` it will find only ASRLCOM1::INSTR

but when i use explicitly `visa.ResourceManager.open_resource("TCPIP0::K-34461A-10709::inst0::INSTR")` it works fine. Same story with other instrument Agilent 34461A.

> Machine Details:
>    Platform ID:    Windows-10-10.0.17763-SP0
>    Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel
> 
> Python:
>    Implementation: CPython
>    Executable:     C:\Users\zapadlik\AppData\Local\Programs\Python\Python36-32\python.exe
>    Version:        3.6.1
>    Compiler:       MSC v.1900 32 bit (Intel)
>    Bits:           32bit
>    Build:          Mar 21 2017 17:54:52 (#v3.6.1:69c0db5)
>    Unicode:        UCS4
> 
> PyVISA Version: 1.9.1
> 
> Backends:
>    ni:
>       Version: 1.9.1 (bundled with PyVISA)
>       Binary library: Not found
>    py:
>       Version: 0.3.1
>       ASRL INSTR: Available via PySerial (3.4)
>       USB INSTR: Available via PyUSB (1.0.2). Backend: N/A
>       USB RAW: Available via PyUSB (1.0.2). Backend: N/A
>       TCPIP INSTR: Available
>       TCPIP SOCKET: Available
>       GPIB INSTR:
>          Please install linux-gpib to use this resource type.
>          No module named 'gpib'

MatthieuDartiailh, 2019-04-09:
This is a known limitation of the pyvisa-py backend (cf https://github.com/pyvisa/pyvisa-py/issues/165). If you have time to work on a fix, you are welcome.