pyvisa/pyvisa #224: Connecting New Focus Venturi Laser Via Ethernet

anthonyjrizzo, 2016-07-05:
Hello,

I'm trying to communicate with a computer-controllable laser via ethernet connection using pyvisa. 

- I am able to ping the IP address of the laser and receive a response from command line

- I am able to connect to the raw socket using NI Max

- The device does not appear when using the command rm.list_resources(), however, I seem to be able to connect to it using newfocus = rm.open_resource("TCPIP0::xxx.xxx.xxx.xxx::xxx::SOCKET") as specified in NI Max

- I can write commands without receiving an error message, but I receive a timeout error when I try to read or query

- I have tried fiddling with different combinations of \r, \n, etc. as both read and write terminations to no avail.

I am looking for help in simply sending an '*IDN?' query and receiving a response. I'd appreciate any help that anyone could give. I am attaching my code that I have written thus far, along with the user manual for the laser which includes somewhat useful information about computer commands beginning on page 30.

Best,
Anthony

[6600-User-Manual-v4.pdf](https://github.com/hgrecco/pyvisa/files/348313/6600-User-Manual-v4.pdf)

import visa

rm = visa.ResourceManager()

newfocus = rm.open_resource("TCPIP0::169.254.158.210::139::SOCKET")

newfocus.write_termination = '\r\n'
newfocus.read_termination = '\r\n'

newfocus.query('*IDN?')

MatthieuDartiailh, 2016-07-05:
According to the manual you can set the termination characters using the front panel of the instrument. Be sure that your settings does indeed match the instrument settings.
Can you communicate using NI max ?
rm.list_resources only list some instruments by defaults (typically not USB RAW and TCPIP SOCKET, have a look at the documentation for the option to list those also).

anthonyjrizzo, 2016-07-05:
I have the termination characters set to CR/LF, so \r\n should be working. Unfortunately, I'm getting the same timeout error. In NI Max, I am able to connect to the device under network devices, and the device status says that it is working properly. In the VISA test panel, I am able to write commands, but when I read or query I get a timeout error. When I am writing the commands I am still including \r\n and the front panel is set to CR/LF so I'm not sure why it isn't able to communicate. Thanks!

MatthieuDartiailh, 2016-07-05:
If things do not work with NI Max, even with proper message ending then I cannot really help. Is there any log on the instrument or any proof that a message is received (try sending a command changing the state to see if it is a write issue or a read issue). You can also try to contact the manufacturer.

MatthieuDartiailh, 2018-02-25:
Closing feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh