pyvisa/pyvisa #225: VI_ATTR_TERMCHAR_EN not working

peendebak, 2016-07-07:
When using pyvisa to read data from an instrument `VI_ATTR_TERMCHAR_EN` is not working properly. I want the data to be read as raw data (i.e. ignore all termination characters). I use  `read_raw` from `pyvisa/resources/messagebased.py` for reading, but the data returned always stops if the character that is set in `visa.read_termination` is encountered.

I can change the read termination character (using  `VI_ATTR_TERMCHAR` in the visa lib), this works fine. However, setting `VI_ATTR_TERMCHAR_EN` to 0 does not have the desired effect of ignoring termination characters.

Some more info:
```
In [7]: v
Out [7]: <'SerialInstrument'('ASRL1::INSTR')>
In [8]: v.visalib
Out [8]: <VisaLibrary('c:\\Windows\system32\visa32.dll')>
In [9]: v.visalib.get_attribute(v.session, pyvisa.constants.VI_ATTR_TERMCHAR_EN)
Out [9]: (0,<StatusCode.success: 0>)
```

MatthieuDartiailh, 2016-07-21:
According to the documentation it is expected that read_raw stop at the termination character (it simply does not strip it). Have you tried setting the read_termination to None ? 

@hgrecco mentionned that a read_all method could be added in #190.

@hgrecco going through the code I think that read_values should be re-formatted to look more like its query counterpart as otherwise for ieee block it will fail.

eendebakpt, 2016-07-22:
@MatthieuDartiailh I set the `read_termination` to `''`, which in  https://github.com/hgrecco/pyvisa/blob/master/pyvisa/resources/messagebased.py#L162
makes it set `VI_ATTR_TERMCHAR_EN` to False. I believed that this should make the read_raw continue at any character.
In the visa docs (http://www.ni.com/tutorial/4256/en/#toc2) I see that this property is _ignored_ depending on `VI_ATTR_ASRL_END_IN`. I will close the issue, but it might be good to add this to the docs of `pyvisa.read_raw`

MatthieuDartiailh, 2020-09-06:
Addressed in 2981b365

closed 2020-09-06 by MatthieuDartiailh