pyvisa/pyvisa #226: 'module' object has no attribute 'IntEnum'

christofferTP, 2016-07-08:
pip list:
```

enum (0.4.6)
enum34 (1.1.6)
PyVISA (1.8)
```

trying to import visa gives the following error: 

```

Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (
Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "c:\Python27\lib\site-packages\visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants
  File "c:\Python27\lib\site-packages\pyvisa\__init__.py", line 45, in <module>
    from .highlevel import ResourceManager
  File "c:\Python27\lib\site-packages\pyvisa\highlevel.py", line 22, in <module>
    from . import constants
  File "c:\Python27\lib\site-packages\pyvisa\constants.py", line 599, in <module>
    class AccessModes(enum.IntEnum):
AttributeError: 'module' object has no attribute 'IntEnum'
```

I have tried reinstalling both enum and enum34  and pyvisa . .I dont know what to do.. I can clearly see the IntEnum function in enum.py..

osama989, 2016-10-14:
did you solve this problem? I am trying to run Pyvisa on python 2.6 and I am having the same issue?

christofferTP, 2016-10-14:
Yes, there's a collision when using both enums. I'd say with 90% certainty that enum34 us backwards compatible. Try uninstalling both and reinstalling only 34

christofferTP, 2016-10-14:
The problem was that enum was earlier in the PATH than 34, making it try that one first,  when it needed 34 functionality. Try :

>>> import sys
>>> print sys.path

osama989, 2016-10-14:
I did this, and now I can import the visa library through no error. But then when I try to test using
rm = visa.ResourceManager()

it gives me an error. 

OSError:could not open visa library

I am trying to run Pyvisa on a relatively old machine (Linux scientific 6.5) and python version is 2.6.x

christofferTP, 2016-10-18:
Did you install the visa drivers from national instruments (for example) ? 

http://pyvisa.readthedocs.io/en/stable/getting_nivisa.html#getting-nivisa

osama989, 2016-10-18:
Thank you, I solved the problem some days ago, and it was as you suggested.

closed 2020-09-06 by MatthieuDartiailh