pyvisa/pyvisa #227: Keithley DMM 7510 pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

hsghori, 2016-07-12:
Hi,

I'm trying to implement the example on page 55 in the Keithley DMM7510 manual
http://ekomess.de/media/import/dmm7510/dmm7510_bedienungsanleitung.pdf

However I keep running into a timeout error

`Traceback (most recent call last):
  
File "test.py", line 40, in <module>
    v = float(dmm.query(':READ?'))

  File "C:\Users\xxxx\Anaconda2\lib\site-packages\pyvisa\resources\messagebased.py", line 407, in query
    return self.read()

  File "C:\Users\xxxx\Anaconda2\lib\site-packages\pyvisa\resources\messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)

  File "C:\Users\xxxx\Anaconda2\lib\site-packages\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)

  File "C:\Users\xxxx\Anaconda2\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))

  File "C:\Users\xxxx\Anaconda2\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed..`

using the python code 

`import visa

rm = visa.ResourceManager()

dmm = rm.open_resource(usb_address, read_termination = '\r', write_termination ='\r')

dmm.write('*RST')

dmm.write(':SENS:FUNC "VOLT:DC"')

dmm.write(':SENS:VOLT:RANG 1')

dmm.write(':SENS:VOLT:INP AUTO')

dmm.write(':SENS:VOLT:NPLC 10')

dmm.write(':SENS:VOLT:AZER ON')

dmm.write(':SENS:VOLT:AVER:TCON REP')

dmm.write(':SENS:VOLT:AVER:COUN 100')

dmm.write(':SENS:VOLT:AVER ON')

v = float(dmm.query(':READ?'))

print v`

I've tried adding a timeout parameter and changing the termination characters to '\n' and '\r\n' with no luck. The code works sporadically - maybe three or four times in a row - but is never consistent and i eventually always end up with the above error. 

Any ideas would be greatly appreciated.

Roanmh, 2016-07-15:
Hello,

I am merely a user of the project, but I have had experience with this error. I receive it when the instrument does not send any data (usually through a fault in my code, e.g. asking for data from a command tat does not push to the output buffer). This could be an issue with the interaction with the device specific, especially if you are sure of the termination character. As for the inconsistency, I am unsure what would cause that. The one thing I can think is that the state of the instrument may not be consistent between your runs,  I see that you reset the instrument but is there anything else that could carry over? Also, i work with GPIB, which has been very reliable, could it be an issue with USB?

Just some thoughts for a fellow user. Again, I don't have any expertise in the library (or really in VISA communications.instruments, haha). Good luck.

ethall, 2016-08-22:
Hi @hsghori , are you still having this issue?

If you are, then what is your pyvisa version and the firmware version of your 7510?

Fapostolache-source, 2020-03-20:
Hello,

I'm having the same issue trying to command a Programmable power supply 
Unless the NI I\O Trace is up and running 

My code just an example: 

import pyvisa
rm = pyvisa.ResourceManager()
ret1 = rm.list_resources()
inst = rm.open_resource(ret1[0])

inst._read_termination = '\n'
inst.write_termination = '\n'
print (inst.query('*IDN?'))

Tried with write & read 
For writing the instruments receives the command for reading  it's returning the error : 

Traceback (most recent call last):
  File "<input>", line 8, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 613, in query
    return self.read()
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

same for query 
anyone have ideas ? 
I'm running from a Windows 10 with python 2.7

MatthieuDartiailh, 2020-03-20:
Please have a look at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html and if you cannot figure out the issue please post a detailed description of what you tried and the output of the `pyvisa-info` command.

Fapostolache-source, 2020-03-20:
Fixed this issue by adding a delay  
print (inst.query('*IDN?',delay = 1))

Thanks for your support

MatthieuDartiailh, 2020-03-20:
Happy I could help !

closed 2020-03-20 by MatthieuDartiailh