pyvisa/pyvisa #228: VISA Shell fails on list command with resources available

Roanmh, 2016-07-12:
Upon executing the list command, an uncaught exception is thrown, crashing the shell. See below for a list of resources gained through programmatic use of PyVISA. The Error and stack trace is shown below. The environment is Windows NT4.0 (outdated yes but I am submitting in-case bug also effects other systems) running Python 2.6 and PyVISA 1.8. I can also provide more information upon request.

**Custom program showing current available resources:**
[image]

---

**Failing run of VISA Shell (also shows a degree of pwd independence):**
[image]

---

**Additionally, use of the open command with an explicit resource name gives the message of the same error (without a fatal failure, however):**
[image]

---

**Note:** If anyone has insight as to this issue, I would be very interested to hear it as I may also put some work into this bug, myself.

Roanmh, 2016-07-13:
After some research I found that this a compatibility issue with Python 2.6. Python 2.6 requires that format strings have indexes in variable placeholders. e.g. `{1}` works, but `{}` will not in python 2.6. Since PyVISA is listed as being compatible with python 2.6, it seems like this is a bug within pyVISA area of support.

@hgrecco Are you interested in fixing this or reviewing a pull request for this issue if I am able to?

hgrecco, 2016-08-22:
Fixed now. Thanks!

closed 2016-08-22 by hgrecco