pyvisa/pyvisa #229: Can't get visa.RessourceManager() working neither with ('@ni') nor with ('@py') arguments

bruno066, 2016-07-23:
Hi, 

after spending some time in reviewing the different possible related causes to my issue I finally decided to post.
I am unable to get any instruments using the RessourceManager() neither with ('@ni') nor with ('@py') arguments.
Here are the related commands and answers.

>>import visa as v
>>v.ResourceManager()             #same output using ('@ni') argument
<ResourceManager(<VisaLibrary('/usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0')>)>
>>rm = v.ResourceManager()
>>rm.list_resources()
()

>>v.ResourceManager('@py')
<ResourceManager(<VisaLibrary(u'unset')>)>
>>rm = v.ResourceManager()
>>rm.list_resources()
VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

I tried this two commands with two different devices:
- one, ethernet connected that works fine with   vxi11.Instruments()
- the other, usb connected, that is identified by lsusb command

I have already installed the latest versions of pyvisa and pyvisa-py:
>>pip install --upgrade pyvisa-py
Requirement already up-to-date: pyvisa-py in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: pyvisa>=1.8 in /usr/local/lib/python2.7/dist-packages (from pyvisa-py)
Requirement already up-to-date: enum34 in /usr/lib/python2.7/dist-packages (from pyvisa>=1.8->pyvisa-py)

Here is the output of a command that can help to solve my problem:
>>python -m visa info
Machine Details:
   Platform ID:    Linux-4.6.0-1-amd64-x86_64-with-debian-stretch-sid
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.12
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Jun 29 2016 08:18:26 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.dev0

Backends:
   ni:
      Version: 1.9.dev0 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.1)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

Hoping my problem has a solution ^^
Thanks for your time.

MrPySmart, 2020-03-19:
Hello as I see there was no solution for this issue since 3.5 years.
I had to struggle with the same/similar issue, but could not find a solution, as nowhere the problem was understood.

Luckily there were some investigations in the internet and I had a suspect that hardend.
Well, the root cause of this issue (at least for my issue) is somehow funny:

1. I found some example Python-script from Keysight/Agilent and as a lot of scripts they start with:
```
import visa
rm = visa.ResourceManager()
```

2. on https://pypi.org/project/PyVISA/ in the section **Requirements** you find:

- Python (tested with 2.7 and 3.4+)
- VISA (tested with NI-VISA 17.5, Win7, from www.ni.com/visa)

So I thought, "just install visa libs" without PyVisa, NI-Visa (National Instruments visa lib) nor PyVisa-py.

THIS is the fault! I search for "visa" on PyPi.org and find https://pypi.org/project/visa/
but this is NOT VISA==Virtual instrument software architecture
it is a lib for PAYMENT with VISA==Credit card ;-)

Maybe on the PyVisa site it should be made clear that on needs a backend for PyVisa like NI-Visa or PyVisaPy.

Solution:
If you do not need "Credit card payments with VISA" uninstall visa
`pip uninstall visa` or `pyhton3 -m pip uninstall visa`
than install PyVisa-Py, which also installs PyVisa.

3. I think there is also a real software-issue with PyVisa:

I uninstalled the VISA-Card visa lib (visa.py) and only installed PyVisa.
Now, when I try to unsinstall PyVisa pip states:
```
python3 -m pip uninstall PyVISA
Found existing installation: PyVISA 1.10.1
Uninstalling PyVISA-1.10.1:
  Would remove:
    /home/dev/.local/bin/pyvisa-info
    /home/dev/.local/bin/pyvisa-shell
    /home/dev/.local/lib/python3.7/site-packages/PyVISA-1.10.1.dist-info/*
    /home/dev/.local/lib/python3.7/site-packages/pyvisa/*
    /home/dev/.local/lib/python3.7/site-packages/visa.py
```

When you install "VISA CARD" visa lib you get a visa folger in /usr/local/lib/python3.7/dist-packages/visa/

Maybe the namings should be distinguished stricter.

MatthieuDartiailh, 2020-03-19:
Thanks for the deep investigation. I actually figured this one out some time ago but forgot to comment on this issue. The visa module of pyvisa is being deprecated and the documentation now only suggests to use directly pyvisa. Sorry for the inconvenience, I fear it will take some time for the resources on the subject

closed 2020-03-19 by MatthieuDartiailh

kariminiaiman1, 2020-09-12:
in [Visa Library](https://visalibrary.com/)
You pick the destination, we’ve done all the research for you! about visa