pyvisa/pyvisa #231: pyvisa can write but cannot read from Keithley 2602A

gzshao, 2016-07-29:
I’m connecting my Keithley 2602B directly to my mac via ethernet.

```
import visa
rm = visa.ResourceManager()
keithley=rm.open_resource(“TCPIP::169.254.000.001::5025::SOCKET")
```

below is my system information
`python -m visa info`

**********************************************
Machine Details:
   Platform ID:    Darwin-15.0.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Users/Guozheng/anaconda/bin/python
   Version:        2.7.11
   Compiler:       GCC 4.2.1 (Apple Inc. build 5577)
   Bits:           64bit
   Build:          Dec  6 2015 18:57:58 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 32, 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
**********************************************

It is totally OK when I write something to Keithley
```
keithley.write("smua.measure.count = 1")
keithley.write("smua.measure.r(smua.nvbuffer1)")
```

I knew that worked because if I go to Keithley recall the value I can see it there.

However, when I tried to read the value by setting
```
keithley.term_chars = '\n'
resist = keithley.ask("printbuffer(1, nvbuffer1.n, nvbuffer1)") 
```

it gives me error message:
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

and on the Keithley display it said:
ERROR CODE:-286
 attemped to index global 'nvfugger1' (a nil value) 

### Another symptom is that when I do:
`keithley.query("*IDN?")`
it gives me this error:
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

I have no idea what that means. Thanks for any suggestion!

gzshao, 2016-08-01:
Using python3.5 cleared the issue.

hgrecco, 2016-08-22:
I am anyway surpirsed becasue I am using python 2.7 without any trouble. Thanks for the information.

closed 2016-08-22 by hgrecco