pyvisa/pyvisa #232: USB Instruments not found Windows 10

mcleung, 2016-08-02:
Hi all,
I'm having trouble "finding" my instruments that are connected via USB.
I know instrument works fine as I can communicate to it via Labview.

About me:
Windows 10 - 64 bit
Python 2.7.12 - Anaconda custom (32-bit)
visa.__version__ = 1.9.dev0'

Trying to connect to Thorlabs CCS spectrometer. When I run:

```
import visa
print("Import success")
rm = visa.ResourceManager()
print(rm.list_resources())
```

The only resources I find are (u'ASRL1::INSTR', u'ASRL10::INSTR')
Yet I know there is a USB resource for my Spectrometer.
The Labview program finds and identifies it as 'USB0::0x1313::0x8087::M00246003::RAW'

Any ideas?

MatthieuDartiailh, 2016-08-10:
By default PyVisa does not list RAW resources, look at the documentation of the list_resources method.

hgrecco, 2016-08-22:
@MatthieuDartiailh I did this to mimick visa.dl but people keep asking about it. So maybe listing only INSTR was a bad choice. What do you think?

MatthieuDartiailh, 2016-08-22:
Given the number of report for that issue it may be worth it, just to reduce the noise in the issues.

juhuntenburg, 2018-04-30:
Hi, I have the same issue of not finding my USB connected device, although it is visible in NI MAX (see screenshot). I have tried rm.list_resources('?*') but no luck. Anything else I can try / might be  doing wrong?

[image]

MatthieuDartiailh, 2018-04-30:
What exactly is your device ? Is it a National Instrument DAQ ? Usually they do not use Visa for communication but National instruments DAQ library that has been in wrapped in Python (see PyDAQmx at https://github.com/clade/PyDAQmx)

juhuntenburg, 2018-05-02:
Ah yes thank you, I go a bit confused there. Thanks for pointing me in the right direction, PyDAQmx should be what I need. Thanks!

closed 2018-05-02 by MatthieuDartiailh