pyvisa/pyvisa #235: Segmentation fault - Ubuntu Bash on Windows

hsghori, 2016-08-11:
Hey,

I recently installed the Windows 10 anniversary update and am trying to get everything configured on the Ubuntu Bash terminal. However, attempting to create a ResourceManager object results in a segmentation fault.
`
$ python
>> import visa
>> rm = visa.ResourceManager('@py')
Segmentation Fault (core dumped)
`
I have pyvisa, pyvisa-py, pyserial,, and pyusb installed - and this happens using Python 2.7 and Anaconda 4.1.1 for Python 2.7.

hgrecco, 2016-08-22:
I guess is a library that it cannot be opened. Try importing pyusb and pyserial.

DavidLutton, 2017-06-18:
[Windows Developer](https://wpdev.uservoice.com/forums/266908-command-prompt-console-bash-on-ubuntu-on-windo/suggestions/15424383-need-serial-usb-com-port-access) site comments suggest that USB devices, like USBTMC,  cannot as yet be accessed. 

There is apparently support for exposing COM ports as /dev/TTYs1
Guide [here](https://blogs.msdn.microsoft.com/wsl/2017/04/14/serial-support-on-the-windows-subsystem-for-linux/) for Windows Subsystem for Linux

MatthieuDartiailh, 2020-09-06:
Closing as there is not much that can be done on PyVISA side.

closed 2020-09-06 by MatthieuDartiailh