pyvisa/pyvisa #236: Query results blank after first time after power up

T-2, 2016-08-31:
New to instrument control as well as Python. Just issuing commands in Python's interactive mode to learn how to talk to a new Keithley power supply.
If I power up the power supply then get into Python and do . . .

>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> instr = rm.open_resource('USB0::1510::8736::9102003::0::INSTR')
>>> instr.query("*IDN?")
u'Keithley instruments, 2220-30-1, 9102003, 1.15-1.04\n'

As you can see that works however if I then repeat the last command I get this ...
>>> instr.query("*IDN?")
u''

I can write to the power supply and set stuff up just fine. It does everything I tell it to do. I can set the voltage and the current, turn the output on and off. I can even write stuff to the display. I just cannot get back results from a query or read unless it is the first such command issued after power up of the power supply. All subsequent requests return u''. I get no error messages either. All the commands that I issue come back with success. Just no results.

I messed with changing the read and write terminators but no change. Anything other then the default terminators came back with errors anyway.

Is this something about the interactive mode of Python? Am I missing something else?
By the way it did the same thing in Linux Mint 17.3 with Python v2.6

Output of  python -m visa info

Machine Details:
   Platform ID:    Linux-4.4.0-21-generic-x86_64-with-LinuxMint-18-sarah
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.12
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Jul  1 2016 15:12:24 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.0.1)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

T-2, 2016-09-09:
I have more info on this.
I found a Siglent scope here that also can be controlled via USB. Set it all up on the same computer, same OS, same version of python, same pyvisa-py and I can talk to it just fine. I get responses every time. No missing responses.

Any ideas?

T-2, 2016-10-30:
Even more info on this:
a) Got a Keithley 3706A and I can talk to it just fine with the same setup. I always get a response.
b) Got another Keithley power supply of the same model and it does exactly the same thing. Same strange one time communication.
c) Switched to python3 and python3-USB and pyvisa3-py and the same thing happens.
d) No errors in any of the system logs either.
e) When it does come back with a correct response there is a slight delay. When it comes back with a blank response it comes back quicker.
f) Instead of query if I do a write then a read I still get the blank response after the first time.

Does "e" above suggest a timing issue? Can I send some command to wait a tad longer for a query response? What do you think?

mligare, 2017-06-30:
I have encountered very similar behavior communicating with a Tektronix TBS 1052B-EDU Digital Oscilloscope.  As in the case of T-2, I can successfully write multiple commands to the scope, and the first read or query is successful.  The next write command, however, results in a  "ValueError: [Errno 110] Operation timed out" (more details in log messages below).  I have tried many combinations of read_termination characters, to no avail (\n, \r, None, \n\r, \r\n) ; some raise warnings, none eliminate the error.  

The only way I have been able to get around this problem is to close communications with the scope after every read/query, and then reopen them; an example of this successful work-around follows the code that fails.  System info follows that.

Code that fails:

>>> import visa
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2017-06-30 15:09:54,084 - pyvisa - DEBUG - SerialSession was correctly imported.
2017-06-30 15:09:54,104 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2017-06-30 15:09:54,109 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name 'TCPIPSession'.
2017-06-30 15:09:54,111 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2017-06-30 15:09:54,111 - pyvisa - DEBUG - Created library wrapper for unset
2017-06-30 15:09:54,111 - pyvisa - DEBUG - Created library wrapper for unset
2017-06-30 15:09:54,112 - pyvisa - DEBUG - Created ResourceManager with session 5286969
>>> scope = rm.open_resource('USB0::1689::872::C019585::0::INSTR')
2017-06-30 15:10:01,970 - pyvisa - DEBUG - USB0::1689::872::C019585::0::INSTR - opening ...
2017-06-30 15:10:02,312 - pyvisa - DEBUG - USB0::1689::872::C019585::0::INSTR - is open with session 5319173
>>> scope.write('*rst')   # Successfully resets scope parameters
(20, <StatusCode.success: 0>)
>>> scope.query('*IDN?')
2017-06-30 15:11:31,826 - pyvisa - DEBUG - USB0::1689::872::C019585::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
'TEKTRONIX,TBS 1052B-EDU,C019585,CF:91.1CT FV:v4.0\n'
>>> scope.query('*IDN?')  # Could be any read, write or query here
.
Traceback (most recent call last):
.
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 202, in write
    return self.usb_send_ep.write(data)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/usb/backend/libusb1.py", line 824, in bulk_write
    timeout)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/remote/anaconda-3.5/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out

Successful work-around, with scope.close(), scope.open() pair inserted:

>>> import visa
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2017-06-30 15:22:42,225 - pyvisa - DEBUG - SerialSession was correctly imported.
2017-06-30 15:22:42,246 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2017-06-30 15:22:42,251 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name 'TCPIPSession'.
2017-06-30 15:22:42,253 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2017-06-30 15:22:42,253 - pyvisa - DEBUG - Created library wrapper for unset
[... 27 lines cut ...]
   Executable:     /usr/remote/anaconda-3.5/bin/python
   Version:        3.5.1
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-1)
   Bits:           64bit
   Build:          Jun 15 2016 15:32:45 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      TCPIP INSTR: Available 
      ASRL INSTR: Available via PySerial (2.7)
      TCPIP SOCKET: Available 
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1

Thanks in advance for any help.

mligare, 2017-07-05:
The behavior I detailed above (only a single `read()` is successful on a Tektronix TBS 1052B-EDU oscilloscope) has also been reported with Keysight devices and a Tektronix TBS1202B-EDU scope; see https://github.com/pyvisa/pyvisa-py/issues/71

Although the work-around described above (closing and re-opening the communications session) works for most commands, it does not work when trying to transfer entire waveforms from the oscilloscope.

T-2, 2017-07-05:
I had to get things done so I gave up on pyvisa and pyvisa-py. I ended up using the older python-usbtmc instead. It does not have the rich feature set as pyvisa but it works every time. Using it for both the Keithly power suppy as well as the 3706A with no problems.

mligare, 2017-07-06:
I have determined in my case that the problem is model specific.  When communicating with a Tektronix TBS 1052B-EDU, successive  `read`/`query` requests fail, but when I test identical 
code with Tektronix TDS 1002B and Tektronix TDS 2024B oscilloscopes everything works.

elkevn, 2018-02-07:
Same problem here with a Thorlabs PM100D powermeter.
Has anyone solved this problem so far? I didn't get into the python-usbtmc solution yet, although i might get to that if it comes to it...

no specific error message in journalctl...

Here are the specs:
`instrument.query("*IDN?")
Traceback (most recent call last):

  File "<ipython-input-90-b54a8d25ab92>", line 1, in <module>
    instrument.query("*IDN?")

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))

ValueError: [Errno 110] Operation timed out`

and here are the infos on pyvisa-py:

`instrument.query("*IDN?")
Traceback (most recent call last):

  File "<ipython-input-90-b54a8d25ab92>", line 1, in <module>
    instrument.query("*IDN?")

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))

ValueError: [Errno 110] Operation timed out`

Thanks in advance for your help!

MatthieuDartiailh, 2018-02-07:
Could you post your full script ? This does not necessary relate to the usbtmc error seen by the others. Have you properly configured the read and write termination characters ?

elkevn, 2018-02-07:
Well, i guess i was wrong, the error is slightly different from other people here; mine comes at the third request, the first one goes through all the way, the second returns nothing, and the third, bingo, error:

```
IPython 6.1.0 -- An enhanced Interactive Python.

import visa

rm=visa.ResourceManager('/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py@py')

ressource = rm.list_resources()[1]

instrument = rm.open_resource(ressource)

instrument
Out[5]: <'USBInstrument'('USB0::4883::32888::P0016534::0::INSTR')>

instrument.query("*IDN?")
Out[6]: 'Thorlabs,PM100D,P0016534,2.5.0\n'

instrument.query("*IDN?")
Out[7]: ''

instrument.query("*IDN?")
Traceback (most recent call last):

  File "<ipython-input-8-b54a8d25ab92>", line 1, in <module>
    instrument.query("*IDN?")

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))

ValueError: [Errno 110] Operation timed out
```

MatthieuDartiailh, 2018-02-07:
Reading your first message it was not clear to me that it was only the third message that produced the error. The termination character you use is indeed the proper one. If you have time to test python-usbtmc, I would be most interested to know if it works for you.
Additionally what happens if you do a read after the first query ? Do you get a timeout as expected ? and if not does reading before querying again fix the issue ? 
I am sorry but I have had little to dedicate to investigating this since I started working on PyVISA so any additional infos you can provide is welcome.

elkevn, 2018-02-08:
Well, truth is, i didn't know either that the error was raised on the third call... sorry about that. 

So interestingly enough, if i just send write commands to the instrument, he does not rise the 110 error.
If i just send read commands though, he goes bad after at the 4th command:

```
instrument.read()
Out[40]: ''

instrument.read()
Out[41]: ''

instrument.read()
Out[42]: ''

instrument.read()
Traceback (most recent call last):

  File "<ipython-input-43-c68ffc95da11>", line 1, in <module>
    instrument.read()

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 253, in read
    return self.sessions[session].read(count)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 102, in read
    usb.USBError)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 313, in _read
    current = reader()

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 96, in <lambda>
    return self._read(lambda: self.interface.read(1),

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 307, in read
    raw_write(req)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))

ValueError: [Errno 110] Operation timed out
```

and if i send query commands, he first gives the correct answer, then the second time he sends back an empty string, and the third time it goes down the tube with an error. (just as in my last message).

same thing with write than read commands: first write/read goes well, second write goes well, the read not exactly as well for it gives back an empty string, and as for the next command, it goes down as an error, whether it be a read or a write: 

with the WRITE as fifth command

```
instrument.write("*IDN?")
Out[97]: (20, <StatusCode.success: 0>)

instrument.read()
Out[98]: 'Thorlabs,PM100D,P0016534,2.5.0\n'

instrument.write("*IDN?")
Out[99]: (20, <StatusCode.success: 0>)

instrument.read()
Out[100]: ''

instrument.write("*IDN?")
Traceback (most recent call last):

  File "<ipython-input-101-d5edac6ce7b9>", line 1, in <module>
    instrument.write("*IDN?")

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))

ValueError: [Errno 110] Operation timed out
```

and with READ as fifth command (not exactly sure how useful this is, but well...)

```
instrument.write("*IDN?")
Out[104]: (20, <StatusCode.success: 0>)

instrument.read()
Out[105]: 'Thorlabs,PM100D,P0016534,2.5.0\n'

instrument.write("*IDN?")
Out[106]: (20, <StatusCode.success: 0>)

instrument.read()
Out[107]: ''

instrument.read()
Traceback (most recent call last):

  File "<ipython-input-108-c68ffc95da11>", line 1, in <module>
    instrument.read()

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 253, in read
    return self.sessions[session].read(count)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 102, in read
    usb.USBError)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 313, in _read
    current = reader()

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 96, in <lambda>
    return self._read(lambda: self.interface.read(1),

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 307, in read
    raw_write(req)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))

ValueError: [Errno 110] Operation timed out
```

I didn't get time yet to try operating with usbtmc, i'll let you know what happens then. If you get any idea of where this comes from ... it's apparently a bug in the read function or with the link with my instrument...

thanks a lot for your help in any case!

MatthieuDartiailh, 2018-02-08:
Thanks for the detailed response. In the first case, what command did you send before the read ? Also what happens if you write then read three or four times in a row ? What I don't get is why we get empty strings it looks like the termination character is somehow not removed from the buffer on first try and is read again after. I hope to be able to dig a bit more into it tomorrow.

MatthieuDartiailh, 2018-02-08:
Also by curiosity does your instrument show any error on its screen when you issue the second write ? Wondering because some instruments complain when one sends a message before reading fully the output of the previous one.

elkevn, 2018-02-08:
in every one of the above cases, i first closed the instrument from last error, then opened it again but that's it. basically i started every code i put from a clean "open(ressource)", so the starting point is the same each time basically. 

About the write then read few times in a row, that's what i was describing in the last bit of my previous message; basically the second read returns an empty string, and then that's it, the next command whichever it is goes into an error. i've tried a few commands, they all terminate with an error if there were two reads..

the instrment does not show any sign of an error occuring, it's being steady as a rock.

Thanks so much for your help, hope you figure it out and my problem might help to it!

ClundXIII, 2018-03-23:
Hi,

I think I am running into the same problem. Using a MSO2002B from tektronix.

`inst.query('*IDN?')` fails after I run it the second time though.

Simon

MatthieuDartiailh, 2018-03-23:
Hi, 
Sadly I never managed to find time to investigate those USBTMC issues in details. Could you in order:
- check if things works with python-usbtmc ?
- try to understand what we do differently and why it leads to this bug ?
Also could you post the output of `python -m visa info` ?

ClundXIII, 2018-03-23:
usbtmc works.
```
# python3 -m visa info
Machine Details:
   Platform ID:    Linux-4.10.0-42-generic-x86_64-with-LinuxMint-18.2-sonya
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.5.2
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Nov 23 2017 16:37:01 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.0.dev0

Backends:
   ni:
      Version: 1.10.0.dev0 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.3.dev0
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

ClundXIII, 2018-03-23:
Edit usbtmc only works for querying once :(

closing and getting a new instrument does not solve the problem. Only plugging the oszi out and in again does the trick.

elkevn, 2018-03-23:
I didn't know that at first, but as long as i am using linux, there is another option to communicate with the instrument:

write (and read) directly from the node! in my case, the node is /dev/usbtmc0
it is instanciated by usbtmc module when i plug it in, and although i don't compltely understand why, it was not called up on my old distro (ubuntu 16.04, although it was on another computer with the same settings...) but it is now on Debian 9.
anyhow, then it's just a matter of opening it in python and writing and reading directly from the file.

e.g.
```
class usbtmc:
        def __init__(self, device):
                self.device = device
                self.FILE = os.open(device, os.O_RDWR)

        def write(self, command):
                os.write(self.FILE, command.encode("utf-8"));

        def read(self, length = 4000):
                return os.read(self.FILE, length)
```

so that now you can just call  
`usbtmc("/dev/usbtmc0")  
or something like this to be able to write directly into the node file. that's how i got aroud the problem anyways...

however, i must say it still bothers me that if i end up reading from the file without it containing anything (by that, i mean if i read and it shouldn't contain anything because i didn't write a correct command before), then it's back to the same problem as before and it never shows anything until i quit the program and it ends up with me having to simply disconnect the device and reconnect it. I don't have access to the specific error right now though, so that last bit isn't exactly helpful in any sense, but just take it as a "warning"...

masbab, 2019-02-18:
@elkevn
I am uisng python_usbtmc in centos 6.10 and having trouble accessing Agilent DSA. Do I need to install usbtmc drivers? 
Here is my system info
python -m visa info
Machine Details:
   Platform ID:    Linux-2.6.32-754.9.1.el6.x86_64-x86_64-with-centos-6.10-Final
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.6.6
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-17)
   Bits:           64bit
   Build:          Aug 18 2016 15:13:37 (#r266:84292)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
Thanks in advance!

elkevn, 2019-02-19:
> @elkevn
> I am uisng python_usbtmc in centos 6.10 and having trouble accessing Agilent DSA. Do I need to install usbtmc drivers?
> Here is my system info
> python -m visa info
> Machine Details:
> Platform [...]

Hey masbab, 
I have absolutely no idea, sorry about it. I suppose you could try the same trick that worked for me though? trying to open the /dev/usbtmc0 node directly as a read-write file? (i did not write that in my previous exemple, but you should import the module named "os" before writing the exemple I gave)

good luck there!