pyvisa/pyvisa #241: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

dishantarora, 2016-10-18:
Hi,

I am trying to use pyvisa for connecting Keysight's device on my localhost. Following is my script but I am continuously getting error

`import visa

rm = visa.ResourceManager('@sim')

rm.list_resources()
instrument = rm.open_resource('TCPIP0::localhost::inst0::INSTR')
instrument.timeout = 4000
print(instrument.query("*IDN?"))
instrument.write(':JITT:HEF:PER:AMPL? "M1.DataOut1"')
print(instrument.read())`

Error:
"C:\Program Files (x86)\Python27\python.exe" C:/Users/disarora/PycharmProjects/untitled2/script
Traceback (most recent call last):
  File "C:/Users/disarora/PycharmProjects/untitled2/script", line 8, in <module>
    print(instrument.query("*IDN?"))
  File "C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 407, in query
    return self.read()
  File "C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Program Files (x86)\Python27\lib\site-packages\pyvisa-sim\highlevel.py", line 187, in read
    raise errors.VisaIOError(constants.VI_ERROR_TMO)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Process finished with exit code 1

MatthieuDartiailh, 2016-10-18:
My first guess is that the termination character for either reading or writing is not right. Refer to the manual to find the right characters (you can set them using .write_termination and .read_termination)

dishantarora, 2016-10-18:
I tried that as well. See below

code
`import visa

rm = visa.ResourceManager()

rm.list_resources()
instrument = rm.open_resource('TCPIP0::localhost::inst0::INSTR',write_termination= '\r', read_termination='\r')
instrument.timeout = 4000
print(instrument.query("*IDN?\r"))
instrument.write(':JITT:HEF:PER:AMPL? "M1.DataOut1"\r')
print(instrument.read())`

Error
"C:\Program Files (x86)\Python27\python.exe" C:/Users/disarora/PycharmProjects/untitled2/script
C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py:1225: VisaIOWarning: VI_WARN_CONFIG_NLOADED (1073676407): The specified configuration either does not exist or could not be loaded. VISA-specified defaults will be used.
  ret = library.viOpenDefaultRM(byref(session))
Keysight Technologies,M8070A,DE5250000002,3.4.30031.1
C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\resources\messagebased.py:401: UserWarning: write message already ends with termination characters
  self.write(message)
C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\resources\messagebased.py:407: UserWarning: read string doesn't end with termination characters
  return self.read()
C:/Users/disarora/PycharmProjects/untitled2/script:9: UserWarning: write message already ends with termination characters
  instrument.write(':JITT:HEF:PER:AMPL? "M1.DataOut1"\r')
Traceback (most recent call last):
  File "C:/Users/disarora/PycharmProjects/untitled2/script", line 10, in <module>
    print(instrument.read('\r'))
  File "C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 335, in read
    message = self.read_raw().decode(enco)
  File "C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Program Files (x86)\Python27\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Process finished with exit code 1

MatthieuDartiailh, 2016-10-18:
When setting the read/write termination you should not add them to each read/write call (which you did). I guess you looked up the termination character in the manual, could you try checking that you can communicate using NiMAX, if this fails it means the termination is wrong.

dishantarora, 2016-10-18:
I installed NiMax and tried to connect using SFP(Soft front panel) but I think it is not connecting there as well. In NI VISA interactive control When tried to write command it works fine but when query or read the same command it gives me timeout error.

Also, I am using Keysight bert software. It is not hardware.

gasparka, 2018-11-06:
Had same problem for RIGOL1000 oscilloscope, this one needed `write_termination='\n'`.

MatthieuDartiailh, 2018-11-06:
To elaborate on @gasparka, one thing you need to make sure is that the instrument does detect the end of your message when you write otherwise it will keep waiting, and this will not generate any visible error. The easiest way to is to pick a command that has a visible effect on the instrument front panel and check that something happen.

gasparka, 2018-11-06:
I think we should add something about this to the documentation, i spent stupid amount of time on this because i was pretty sure that '\n' is the default write termination...
Ideas?

MatthieuDartiailh, 2018-11-06:
I just noticed that the docs have some problem currently (wrong tool bar) and I will try to fix it. The docs could use some reorganization. In particular the tutorial, advanced tutorial and writing-reading values could be grouped together. If we do that we could also add a guide on how to proceed for a new instrument. The order would be tutorial, advanced tutorial, new guide, reading-writing values.

ZhaoMoMo, 2019-01-23:
@MatthieuDartiailh

Hello, I use RS232 to communicate with a KISTLER Torque sensor 4700. And I also have the same problem, please help me.
I noticed that you said "The easiest way to is to pick a command that has a visible effect on the instrument front panel and check that something happen." I tried to use NI MAX -> Open VISA test panel to execute a query command and I got a measurement data as reply. but the python script still report VI_ERROE_TMO(-1073807339). My code as below:

import visa

rm = visa.ResourceManager()
TorqueMeter = rm.list_resources()[1]
TorqueInst = rm.open_resource(TorqueMeter)
TorqueInst.timeout = 5000
TorqueInst.read_termination = "\r\n"
TorqueInst.write_termination = '\r\n'
TorqueInst.write("MEAS:TORQ?")
print(TorqueInst.read())

"C:\Program Files\Python37\python.exe" D:/MotorAutoPlatform/VisaFunc/Torquemeter.py
Traceback (most recent call last):
  File "D:/MotorAutoPlatform/VisaFunc/Torquemeter.py", line 10, in <module>
    print(TorqueInst.read())
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\resources\messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\resources\messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Process finished with exit code 1

MatthieuDartiailh, 2019-01-23:
I assume you use the same termination chars than NI-Visa (if not that is the first thing to change). And that you have checked your instrument manual.
Since you are using a serial resource make sure that all parameters (baud_rate, parity etc) match.
If you believe that the write is successful try reading the answer one byte at a time using `read_bytes`. If that fails it means the write did not succeed and you need to find the proper write_termination.

MatthieuDartiailh, 2019-01-24:
Try reading a single byte at a time, this way we will know if the instrument answer.

For the serial, have a look at https://github.com/pyvisa/pyvisa/issues/326

DarioDiem, 2019-05-30:
> @MatthieuDartiailh
> 
> Thank you for your reply. :)
> Yes, I use the same termination chars as NI-visa, and checked with instrument manual.
> Could you please tell me how to set serial parameters with pyvisa, I can't find the method in pyvisa docs.
> My code " TorqueInst.write("MEAS:TORQ?")" being executed without error, can we make sure the write_termination is correct. also consider to the first item, the termination should not be incorrect.

did you try your command as `query` ? I was having a similar issue, some commands can only be sent as `query` not as `write` 
TorqueInst.query("MEAS:TORQ")

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity feel free to reopen if necessary.

closed 2020-09-06 by MatthieuDartiailh