pyvisa/pyvisa #242: How to set flow control for serial device revisited

magnehh, 2016-10-26:
https://github.com/hgrecco/pyvisa/issues/128

The method in the link above no longer works apparently, and I can't seem to reopen that issue, so here's a new one.

I get the "NameError: name 'constants' is not defined" error.

I know there's an instr.flow_control parameter, but what are the possible arguments?

DavidLutton, 2017-06-18:
I don't have any serial instruments, with me, to try this out
These appear to be the available/relevant constants & #128. 
[pyvisa.constants](http://pyvisa.readthedocs.io/en/stable/_modules/pyvisa/constants.html?highlight=ASRL_flow) 

if you are opening a resource this, is where you could set it [SerialInstrument.flow_control](http://pyvisa.readthedocs.io/en/stable/api/resources.html?highlight=flow#pyvisa.resources.SerialInstrument.flow_control)

```
from visa import constants

constants.VI_ASRL_FLOW_NONE
constants.VI_ASRL_FLOW_XON_XOFF
constants.VI_ASRL_FLOW_RTS_CTS
constants.VI_ASRL_FLOW_DTR_DSR
```

magnehh, 2017-07-05:
Excellent! It seems to work now. Thank you! :-)

closed 2017-11-05 by MatthieuDartiailh

baettigp, 2025-09-24:
Hi Matthieu

Thank you, I got it to work in the meantime. My code is:
`
rm = pyvisa.ResourceManager()
#('@py')
rm.list_resources()
print(rm.list_resources())

print("ADC")

ADC = rm.open_resource("COM3")
ADC.flow_control = VI_ASRL_FLOW_DTR_DSR

#No flow control	VI_ASRL_FLOW_NONE
#RTS/CTS	VI_ASRL_FLOW_RTS_CTS
#DTR/DSR	VI_ASRL_FLOW_DTR_DSR
#XON/XOFF	VI_ASRL_FLOW_XON_XOFF

ADC.baud_rate = 57600
ADC.write_termination = '\n'

print(ADC.query("*IDN?"))
`

I hope this helps someone else struggling to use an arduino/raspberry pi pico (with [Vrekrer SCPI](https://github.com/Vrekrer/Vrekrer_scpi_parser)) with pyvisa without soldering a capacitor to the reset pin. 
Thank you very much
Pio