pyvisa/pyvisa #243: Is it possible to set up events for when devices are connected?

roboguy222, 2016-11-01:
I am under the impression that when a USB, serial, or GPIB device is plugged into a computer, the computer and the device go through some sort of handshake and the device is assigned an address. It seems like I should be able to set up an event to trigger on this, and do some initialization action whenever a device is plugged in.  I have not been able to find any documentation about pyvisa events, however. How do I set up such an event?

I know I could use list_resources() to check what devices are available every second or something, but this seems hacky. Does listing the resources actually query the resources, or just list the ones it already knows?

mbirthelmer, 2016-11-01:
Udev does this for usb (assuming Linux - other OSes will have their own
facilities for this purpose to load drivers, launch applications, etc.).

Serial (rs232) doesn't have that - the devices need to be configured
correctly before any communication can take place, and there is no standard
way for this to happen automatically.

roboguy222, 2016-11-01:
OK, so visa does not do any of this?  How does it know what resources are connected?  Do all the buses send out queries to the hardware whenever `list_resources()` is used, or is this a list that already is in memory and is just accessed?

mbirthelmer, 2016-11-01:
For USB the OS provides services to list devices that advertise themselves
as belonging to a specific class (tmc is the class that test instruments
belong to).

Additionally, pyvisa may query serial ports and specific types of gpib
adapters that it knows about, but ultimately it's up to the OS to provide
these facilities.

Pyvisa is an API that can interface to these devices, but it's up to the OS
to make them available (when possible).

roboguy222, 2016-11-01:
Thank you.  So whenever pyvisa is asked to `list_resources()`, it just asks the OS to list the resources and passes them along?

abhilashreddyp, 2016-12-06:
Hi roboguy222,
Did you tried to implement this functionality. Please let me know.
Thankyou.

roboguy222, 2016-12-06:
I did not try very hard.  Asking pyvisa to `list_resources()` seemed to take longer depending on what size array it was, which surprised me.  If you try this I would be really interested to see how you accomplish it!

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity and also because this is not a capability exposed by the VISA library itself.

closed 2020-09-06 by MatthieuDartiailh