pyvisa/pyvisa #245: Downloaded the latest PyVisa 1.8 [MS says its a VIRUS!!]

bhensley86, 2016-11-17:
I just downloaded PyVisa 1.8 from https://pypi.python.org/pypi/PyVISA and within a few seconds of downloading it I got an error message from Windows Defender on Windows 10 saying it was a Virus labeled "**Trojan:Win32/Spursint.A!cl**" and was labeled as Severe from Microsoft. Whats going on guys?

hgrecco, 2016-11-17:
It is a false positive. The zip file is completely safe. You can upload the file to any of the online virus scanner to check. 

Feel free to reopen.

closed 2016-11-17 by hgrecco

bhensley86, 2016-11-17:
I tried downloading again and got the same result from windows defender. Coworker tried it, and had no issues. Then I tried it again and it was fine. Really weird!

Brian