pyvisa/pyvisa #246: how to use PyVisa API in C++

prakashmnnit81, 2016-12-01:
Is it possible  to use PyVisa API in C++? if yes then How?

I am  able to use PyVisa in Python  but struggling to use pyvisa API in C++

mbirthelmer, 2016-12-01:
Why not use the native VISA libraries? See e.g.
http://www.keysight.com/upload/cmc_upload/All/IO_Libraries_Suite_15_visa.pdf?&cc=US&lc=eng
. I'm sure you can find similar guides for Linux if that's what you want.

Pyvisa is a Python API around a VISA back-end (either the
architecture-native NI VISA libs or py-pyvisa which is a native python
implementation). If you want to use something other than python, there is
no reason to use pyvisa.

prakashmnnit81, 2016-12-02:
Thanks for Reply.   Actually  am C++  programmer  and I am having very Little knowledge of Python  since Python is also providing C++  Integration  so I am wondering  why pyvisa is not providing  C++ integration

mbirthelmer, 2016-12-02:
Maybe I am misunderstanding your question. If you use C++ and you want to
use VISA to control test instruments, you can use the NI (or other) VISA
libraries directly from C++, as normal libraries.

If you want to call C++ libraries from Python, it can be done as described
here: https://docs.python.org/2/extending/extending.html .

But if you are already working in C++, and you don't have a need to use
Python, there is no reason to use pyvisa - just use the VISA libraries
natively.

prakashmnnit81, 2016-12-02:
thanks for reply. actually my  objective behind that since pyvisa is providing implementation independent irrespective of which  visa library  is installed, whether it is NI visa or keysight visa

mbirthelmer, 2016-12-02:
I understand now. Have you looked at this document?
http://www.keysight.com/upload/cmc_upload/All/knowledge_code_02_dual_visa_readme.pdf

prakashmnnit81, 2016-12-02:
I quickly went through this document  so how pyvisa  is handling multiple visa libraries?

closed 2017-11-05 by MatthieuDartiailh

arr-ee, 2024-07-08:
You get header files by installing a vendor’s VISA implementation. You cannot use pyvisa from C++.

See https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019KZcSAM&l=en-GB & NI-VISA user manual linked from there to get started.