pyvisa/pyvisa #253: OSError: Could not open VISA library

piyushgohel, 2016-12-21:
Hello,

I am using visa library in my python script. But it will give following error.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/vrs/Piyush/Task/19_dec/local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1488, in __new__
    visa_library = open_visa_library(visa_library)
  File "/home/vrs/Piyush/Task/19_dec/local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1460, in open_visa_library
    return cls(argument)
  File "/home/vrs/Piyush/Task/19_dec/local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 96, in __new__
    raise OSError('Could not open VISA library:\n' + '\n'.join(errs))
OSError: Could not open VISA library:

I am using
         1.  Ubuntu 14.04
         2.  Python  2.7

wsande, 2017-03-03:
I'm getting the same error in Windows, Python 2.7. 

```
File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 96, in __new__
    raise OSError('Could not open VISA library:\n' + '\n'.join(errs))
OSError: Could not open VISA library:
```
See also Issue 258 with full details:
[https://github.com/hgrecco/pyvisa/issues/258](url)

MatthieuDartiailh, 2018-02-25:
Closing feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh