pyvisa/pyvisa #257: Extra . added after a TCP based VISA write using V 1.8

camerin, 2017-02-21:
When i execute the following code to query a device id on a socket based  VISA connection:
`
import visa
tempInstance = rm.open_resource(""TCPIP0::192.168.100.254::24000::SOCKET")
instIDN=tempInstance.query('*IDN?')
`
I actually see transmitted to the visa instrument is "*IDN?." using a visa connection monitor in keysite connection wizard.
[image]

dvincentwest, 2017-02-21:
Wow that's weird. If you change the write termination does it change the outcome?

camerin, 2017-02-21:
If I send a write then read it is exactly the same. I have used USB instrument in the past with no issues like this

dvincentwest, 2017-02-21:
I mean what happens if you change the string value for tempInstance.write_termination. For that matter can you check what the value is set to?

camerin, 2017-02-21:
I will check when I get to the instrument.

MatthieuDartiailh, 2018-02-25:
I have seen no such things in my recent testings. Closing feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh