pyvisa/pyvisa #259: VI_ERROR_TMO: Timeout expired before operation completed.

b27aj8s6, 2017-04-06:
Hi,

I'm trying to control a Keithley 2410 via python on an x86_64 linux machine (CentOS 7). I'm using python 2.7.5, NI-VISA 16 and PyVISA 1.8. The Keithley is connected to the computer via a a RS232-to-USB cable and it's configured for RS232 communication (with the parameters: baud 57600, bits 8, parity none, terminator <LF>, flow-ctrl none). When I try to query the Keithley I get a timeout error, as shown below:

>>> import visa
>>> rm = visa.ResourceManager()
>>> print(rm.list_resources())
(u'ASRL1::INSTR', u'ASRL2::INSTR')
>>> keithley = rm.open_resource("ASRL2::INSTR")
>>> print(keithley.query('*IDN?'))
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
File "/usr/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 407, in query
return self.read()
File "/usr/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 332, in read
message = self.read_raw().decode(enco)
File "/usr/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
chunk, status = self.visalib.read(self.session, size)
File "/usr/lib/python2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1582, in read
ret = library.viRead(session, buffer, count, byref(return_count))
File "/usr/lib/python2.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Any suggestions as to what the problem(s) might be would be much appreciated!

Thanks

dvincentwest, 2017-04-19:
are you able to test this connection with NI-VISA interactive control?  Most likely this is not a pyvisa "issue" but rather a connection configuration issue.  If you are able to query the instrument with NI tools but not with pyvisa, that might indicate a problem.

There are lots of other things to check.  Are you certain that ASRL2 is the right connection for the instrument?  Are you certain the instrument itself is operating properly?  Can you connect on Windows without a problem?  Etc.

b27aj8s6, 2017-05-03:
Thanks for the suggestions but it seems as though I needed to explicitly set the baud rate.

rm = visa.ResourceManager()
smu = rm.open_resource("ASRL2::INSTR")
smu.baud_rate = 57600

closed 2017-05-03 by b27aj8s6

mohdrafik, 2023-11-27:
you need to add a line for the termination: like this : 
keithley.write_termination = '\r' 
 then it will work