pyvisa/pyvisa #261: AttributeError: 'VisaLibrary' object has no attribute 'viParseRsrcEx'

schaefer0, 2017-05-12:
Just curious, has this ever been fixed from 2014?
I'm getting this error in 2017

MatthieuDartiailh, 2017-11-13:
Which version are you using and what command leads to this issue ?

schaefer0, 2017-11-15:
Hi, Matthieu

From my notes: v1.8-2

 this was back in May 2017, I gave up switched to TCP/IP.

bob s.

ClundXIII, 2017-11-28:
I am getting the same error when trying to access an oscilloscope. Both in 32 and 64 bit python3.

pyvisa version 1.8

MatthieuDartiailh, 2017-11-28:
What VISA implementation are you using (National Instruments, Keysight, Tektronic, ...) ? This function should be implemented by the VISA library and I guess is implemented by at least some implementations as otherwise it would never work.

schaefer0, 2017-11-28:
Googling the error message (months ago so my memory is hazy) I found that an earlier version worked but the attribute was removed in later versions.

ClundXIII, 2017-11-28:
I am not sure. I have somewhat both installed, however the national instruments one only provides 32 bit libraries for Linux.
I will try to remove it and force the opensource package manager version. (I am on Linuxmint 18.2)

MatthieuDartiailh, 2017-11-29:
All specifications I have found about VISA say that viParseRsrcEx should be implemented, which is why this issue surprises me.

ClundXIII, 2017-11-29:
I have the following packages installed:
```
ii  libvisa-dev:amd64                                           0.0.20130812-2                                           amd64        Virtual Instrument Software Architecture (development files)
ii  libvisa0:amd64                                              0.0.20130812-2                                           amd64        Virtual Instrument Software Architecture (runtime files)
ii  python-pyvisa                                               1.8-2                                                    all          Python bindings for Virtual Instrument Software Architecture
ii  python-pyvisa-py                                            0.2-1                                                    all          Backend that implements a large part of the VISA in pure Python
ii  python3-pyvisa                                              1.8-2                                                    all          Python 3 bindings for Virtual Instrument Software Architecture
ii  python3-pyvisa-py                                           0.2-1                                                    all          Backend that implements a large part of the VISA in pure Python 3
```

(I specified the 64 bit library explicitly in the resource manager command to make sure I am using the correct one.)

MatthieuDartiailh, 2017-11-29:
Apparently librevisa does not implement viParseRsrc (and neither viParseRsrcEx) (see here https://github.com/charlylima/LibreVisa/blob/master/README.md). I am not sure how to get around this.

GyrosGeier, 2017-12-03:
I left that function out initially because all the Visa client code I had didn't use it. It should be trivial to implement, so while I'm sick in bed I can give it a poke.

MatthieuDartiailh, 2017-12-03:
Sounds good  @GyrosGeier

MatthieuDartiailh, 2018-01-15:
any progress on this @GyrosGeier

MatthieuDartiailh, 2018-02-25:
ping @GyrosGeier

GyrosGeier, 2018-09-09:
I've found a few minutes to work on this over the weekend.

Apparently I'd written the backend infrastructure for viParseRsrc/viParseRsrcEx back in 2012, but never connected it to the frontend methods, or implemented it in the transport plugins.

Still needs some work, as we have lots of duplicate code between parse/find/open, and I'd like to get rid of that before turning it into a release. In the same run, I'd also like to get a few more attributes supported (especially where the documentation for viParseRsrc says that return values need to correspond to attribute values).

MatthieuDartiailh, 2020-09-06:
@GyrosGeier did you ever complete this work ?

Closing the issue since PyVISA is behaving as expected here. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh

kariminiaiman1, 2020-09-12:
[visa library](https://visalibrary.com/) is a site about visa and passport

ben-lei, 2020-11-20:
@GyrosGeier, has this been addressed? I'm experiencing the same behavior

adsherman09, 2024-07-26:
This is still happening to me in 2024, trying to connect to a USB device from an ubuntu computer.