pyvisa/pyvisa #262: Threading help and thread safety in pyvisa

swirhun, 2017-05-19:
Hi there,

This question relates to multiprocessing.Process and pyvisa.

Can someone help explain how to pass opened resources between threads? Doing it in the obvious way (where I think multiprocessing.Process just pickles the object when the thread is created) seems to cause a VisaIOError. Basically, I have a "main thread" where the instruments are instantiated. I pass them to a sub thread where a "handler" function is spinning, waiting for packets over a socket connection. When a packet is received, the handler issues some command to the instrument. This doesn't work, but interestingly after it fails, I can still control the instrument from the main thread.

Here's a trace of the error:

  File "XYZ.py", line 195, in write
    self._inst.write(':OUTP OFF')
  File "/Library/Python/2.7/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/Library/Python/2.7/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/Library/Python/2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1864, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "/Library/Python/2.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
VisaIOError: VI_ERROR_CONN_LOST (-1073807194): The connection for the given session has been lost.

I [perhaps naively] thought this would work because pyvisa says it's thread safe. But, I guess I am not passing the instrument resource across the threading boundary correctly. What is the correct way? Should I instead pass some resource string and create a separate object in the sub-thread? The instrument resources have already been opened in the main thread.

swirhun, 2017-05-19:
Maybe I will try opening the instruments with a shared lock, and then passing that access key. There are surprisingly no examples of this online, as far as I can tell. I had assumed that "unlocked" (the default) would allow the object to be passed between threads...

MatthieuDartiailh, 2017-05-19:
Multiprocessing uses different processes not threads hence there is no shared memory. And apparently the pickling unpickling does not reopen the resource (which it can't really do as it would need also the resource manager). You would be better opening the resource directly on your worker. By the way the thread safety simply means that VISA won't crash for order from different threads but there is no lock preventing out of order operations on a single resource. Fir that you need to use a lock manually.

swirhun, 2017-05-19:
Thanks for the help! I will try it out with threading instead of
multiprocessing. This is my first time doing something fairly complex with
those packages.

Paul

closed 2017-05-19 by swirhun