pyvisa/pyvisa #264: Installation of pyvisa using pip in Python script

AntonAnton222, 2017-05-30:
Hi,
I'm working with Python 3.6 for windows 7 32 bits, and I'm trying to make a code who install automatically the python packages not installed yet. I use the command :
pip.main(['install','pyvisa'])
It works for all other packages but not for Pyvisa. When I use the command "pip install pyvisa" it works perfectly. Where the problem comes from ?
Thanks
AF

DavidLutton, 2017-06-18:
Not sure about this, yet, I will have a go soon at making [prepare_venv.py](https://gist.github.com/DavidLutton/cc763a577ae48f686e14a2252a31f019) work for a Python 3.6 on a Windows machine

MatthieuDartiailh, 2018-02-25:
Closing feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh