pyvisa/pyvisa #265: PyVISA has moved

hgrecco, 2017-05-31:
PyVISA and its related project (pyvisa-py and pyvisa-sim) has been moved to its own organization.

dvincentwest, 2017-05-31:
is there a link?

hgrecco, 2017-05-31:
You are there already!

https://github.com/pyvisa/pyvisa

dvincentwest, 2017-05-31:
ahhh! of course!  Thanks very much

MatthieuDartiailh, 2017-05-31:
Who will be participating to the projects and is there a roadmap ?

hgrecco, 2017-05-31:
@MatthieuDartiailh The idea is to turn it into a community project and hopefully you and others would like to join.

Regarding the roadmap, it is a community project so it is to the community to discuss and decide. I think there a few interesting things to move forward like asyncio.

MatthieuDartiailh, 2017-05-31:
Count me in. I can't promise I will make large contributions in the near future but I will happily review any proposed changes (there is one pending PR on pyvisa-py for ethernet that deserves some attention !)

hgrecco, 2017-05-31:
Awesome. By the way, I am using bors ng in pint to merge cleanly. We might give it a try for pyvisa as well

MatthieuDartiailh, 2017-05-31:
Sounds like a reasonable idea to me. I will be waiting for an invitation.

MatthieuDartiailh, 2017-06-21:
Bors seems to work on PyVisa-py (even though it appears a bit slow, by the way how can I check what it is doing). One thing that I find annoying however is not being able to edit the tag of an issue or to close it. There is quite large number of issues that need to be either addressed or closed and tagging would help prioritize. Could you give me those permissions ?

MatthieuDartiailh, 2018-01-25:
This has been effective for several months now so closing.

closed 2018-01-25 by MatthieuDartiailh