pyvisa/pyvisa #266: *STB? handling

aholtzma, 2017-06-05:
Shouldn't a query of '*STB?' be converted into a call to read_stb()? Trying to understand how this works w/o corrupting the message available flag in the status byte.

aholtzma, 2017-06-05:
Or worse, getting output messages in the wrong order.

MatthieuDartiailh, 2017-06-06:
Some instruments do not support read_stb (defined in IEEE488.2) while they support  *STB?, so an automatic conversion is a bad idea. That being said I am not sure how *STB? behave with respect to pre-existing values in the output buffer.

MatthieuDartiailh, 2017-06-21:
Any way an automatic conversion is not desirable, as it could too easily do the wrong thing.
This can be closed.

aholtzma, 2017-06-21:
I think the *STB? command overwriting the output buffer is expected behaviour, as I've seen Rohde & Schwarz example code mention work arounds. There is something called a GPIB serial poll which reads all of the STB registers on the bus w/o using the messaging buffer.

closed 2017-06-21 by aholtzma

MatthieuDartiailh, 2017-06-21:
Thanks for closing @aholtzma