pyvisa/pyvisa #267: Issue with query_binary_values

bjaraujo, 2017-06-07:
I am using query_binary_values and it stops in messagebased.py:

```
            while len(block) < expected_length:
                block += self.read_raw()
```

It doesn't timeout although I have a timeout. It just stops. Initially, I thought I wasn't sending enough data but then I used raw sockets and the data is there. I then pass the data to: 

`util.from_ieee_block(block, 'h')`

and everything works fine. I was trying to build a device using pyvisa-sim but since this is binary data not sure how to do it.

MatthieuDartiailh, 2017-06-14:
Looking quickly at the code it does look like the VISA settings for the termination characters is not modified so I suspect it is indeed expecting a termination char. I have sadly no time to investigate on a real instrument.

bjaraujo, 2017-06-14:
Hi, 

Thanks for the answer. 

I guess it is just not updating the read_termination character so if you open it with read_termination on it will be always expecting that.

I could make the change but I am not sure how to change the code as I am not familiar with pyVISA. I can also test if the change worked...

bjaraujo, 2017-06-15:
Actually I think this might be an issue of the VISA driver so going to close.

bjaraujo, 2018-02-02:
I think the bug is that read_raw only reads up to the termination character even if you specify:

inst.read_termination = None

MatthieuDartiailh, 2018-02-04:
Are you using the ni backend or pyvisa-py ? in the second case could you try again with the latest master ? A number of fixes have been made to the tcpip-socket session. Also please note that in most cases, I have encountered the instrument send a termination character also when querying binary data. Could check the content of the bytes you get from the socket to assert it does not contain the termination.

bjaraujo, 2018-02-05:
@MatthieuDartiailh 

Hi, I am using both. It seems read_raw always stops at byte 0x0A which I guess is the default terminator. 
Thanks for the tip though, I will certainly try the latest pyvisa-py.

MatthieuDartiailh, 2018-02-05:
But even if read_raw stops the while loop should ensure that you get all your data.

MatthieuDartiailh, 2018-02-25:
You can use the `read_bytes` method from PyVISA 1.9 to debug this further.
Closing feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh