pyvisa/pyvisa #269: resource.get_visa_attribute() breaks on certain attributes

drordotan, 2017-06-18:
Hi

I am trying to control my device with visa , and to do it I need to get the value of the VI_ATTR_USB_RECV_INTR_DATA attribute. However, this results in an exception:

 _File "C:\Anaconda2\envs\py27_32\lib\site-packages\pyvisa\resources\resource.py", line 257, in get_visa_attribute
    return self.visalib.get_attribute(self.session, name)[0]
  File "C:\Anaconda2\envs\py27_32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 515, in get_attribute
    attr = attributes.AttributesByID[attribute]
KeyError: 1073693104_

The reason is that although VI_ATTR_USB_RECV_INTR_DATA  is defined in visa.constants, no corresponding class is defined in pyvisa.attributes

Is there a way to solve this, or a work-around?

Thanks
Dror

**Machine Details:**
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 42 Stepping 7, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda2\envs\py27_32\python.exe
   Version:        2.7.13
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          May 11 2017 14:07:41 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5242883
         Spec. Version: 4194304
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5242883
         Spec. Version: 4194304

MatthieuDartiailh, 2017-06-18:
You need to add a definition for this attribute in attributes.py. You should probably look in the VISA documentation what are the allowed values and deduce what base class to use. If you succeed in fixing your issue, please submit a pull request with your fix I will be happy to review and merge it. If you can't I will try to give a look but without an instrument to test it is not ideal.

MatthieuDartiailh, 2020-09-06:
The missing attribute has been implemented, so closing the issue.

closed 2020-09-06 by MatthieuDartiailh