pyvisa/pyvisa #270: Compatibility issues with ironpython and pyvisa

OrangeHot, 2017-07-05:
Now i found some hard questions when using a combination ironpython and pyvisa in my work. I think there maybe compatibility issues with ironpython and pyvisa but i have not found the reason yet because i am a beginner in python. Are there any suggestions about this?

MatthieuDartiailh, 2017-07-05:
The first that comes to my mind is IronPython support of ctypes. Some early version did not support it. Could you please send an example file and a traceback,  ?

OrangeHot, 2017-07-11:
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Xml;
using IronPython.Hosting;// import  IronPython
using Microsoft.Scripting.Hosting;
using System.Windows.Forms; 

namespace _PythonScriptCall
{
    class PythonScriptCall
    {
        ScriptEngine py_engine = null;
        ScriptScope py_scope = null;
        ScriptSource py_source = null;
        public void InitPythonEngine()//Init IronPython Engine
        {

            py_engine = IronPython.Hosting.Python.CreateEngine();//Init IronPython Engine
            py_scope = py_engine.CreateScope();
        }
        public void LoadPythonScript()
        {

            py_source = py_engine.CreateScriptSourceFromFile("CMWCtrl.py");//Create python script file
            py_source.Execute(py_scope);//Here will be a error  shouwed as below

        }
        public void ConnectDevice()
        {
            var get_text = py_scope.GetVariable<Func<object>>("get_text");
            var strT = get_text().ToString();
            MessageBox.Show(strT);
            var lanConn = py_scope.GetVariable<Func<object>>("Lan_connect");
            lanConn();
        }
    }
}

OrangeHot, 2017-07-11:
[image]

OrangeHot, 2017-07-11:
//CMWCtrl.py
import sys
#print(sys.path)
#sys.path.remove(r'D:\Program Files (x86)\IronPython 2.7\Lib')
#sys.path.remove(r'D:\Program Files (x86)\IronPython 2.7\DLLs')
#sys.path.remove(r'D:\Program Files (x86)\IronPython 2.7\Lib\site-packages')

sys.path.append(r'D:\Program Files (x86)\IronPython 2.7')
sys.path.append(r'D:\Program Files (x86)\IronPython 2.7\Lib')
sys.path.append(r'D:\Program Files (x86)\IronPython 2.7\Lib\Dlls')
sys.path.append(r'D:\Python27')
sys.path.append(r'D:\Python27\Lib')
sys.path.append(r'D:\Python27\DLLs')
sys.path.append(r'D:\Python27\Lib\site-packages')
sys.path.append(r'D:\Python27\Lib\site-packages\PyVISA-1.8-py2.7.egg')
sys.path.append(r'D:\Python27\Lib\site-packages\enum34-1.1.6-py2.7.egg')
#print(sys.path)
#import nltk
import visa
import pyvisa
#import logging

#sys.platform = "win32"
#import nltk

def Lan_connect():
    a =1;
    b =2;
    c =a+b;
    remote_addr ='192.168.0.1'
    #print(remote_addr)
    rm=visa.ResourceManager()
    #lanremote = rm.open_resource('TCPIP0::'+remote_addr+'::inst0::INSTR')
    #lanremote = rm.open_resource('TCPIP0::11::inst0::INSTR')
    #lanremote.timeout = 60000.0
    #lanremote.ask('*IDN?')   
    #send_query("*IDN?")
    #print('open_resource over')
def get_text():
    return "text from sayhello.py"

OrangeHot, 2017-07-11:
If  i remove import visa and import pyvisa in CMWCtrl.py file, the error will disappear

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity (which is partially my fault).  Feel free to reopen if necessary, however I have no experience with IronPython and may be able to provide only limited support.

closed 2020-09-06 by MatthieuDartiailh