pyvisa/pyvisa #271: WindowsError: Access violation reading 0x034DD000

thamNTC, 2017-07-17:
Hi, running through the tutorial of pyvisa I can perform the following commands without problems:

>>> import visa
>>> rm = visa.ResourceManager()
>>> rm.list_resources()
('ASRL1::INSTR', 'ASRL2::INSTR', 'GPIB0::14::INSTR')
>>> my_instrument = rm.open_resource('GPIB0::14::INSTR')

Unfortunately, the next command: 
>>> print(my_instrument.query('*IDN?'))

results in an error that I've not yet encountered using Python... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 40
1, in query
    self.write(message)
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 20
7, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 18
5, in write_raw
    return self.visalib.write(self.session, message)
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1864,
 in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
WindowsError: exception: access violation reading 0x034DD000

Any ideas?

MatthieuDartiailh, 2018-01-12:
I missed this issue, sorry. Could you please post the output of `python -m visa info` to help me debug this ?

MatthieuDartiailh, 2018-01-12:
This looks like it may be an issue with some C related size of variable on 32 bits system. As I do not have access to one, I would need somebody able to run some tests on such a platform to debug it.

mandalav, 2018-01-15:
Hi,

thanks for the response. Below is the visa info

Machine Details:
   Platform ID:    Windows-XP-5.1.2600-SP3
   Processor:      x86 Family 6 Model 23 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.11
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          Dec  5 2015 20:32:19 (#v2.7.11:6d1b6a68f775)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5242883
         Spec. Version: 4194304
      #2: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5242883
         Spec. Version: 4194304

Thanks,
Venkat

MatthieuDartiailh, 2018-01-15:
I am rather perplex in front of this, and not being able to test does not help. I notice that you VISA version is rather old, could you try updating it ? Also for my own curiosity could you the following line in https://github.com/pyvisa/pyvisa/blob/master/pyvisa/ctwrapper/highlevel.py#L76: `print(tmp)` ? I would like to know why you get the same lib listed twice.
I will try to investigate further, but it may take some time do not hesitate to ping me.

mandalav, 2018-01-15:
> ctwrapper/highlevel.py#L76: print(tmp) ? I would like to know why you get

Thanks again. Is the latest VISA version compatible with python 2.7? We
have many scripts developed in python 2.7.

Thanks,
Venkat

MatthieuDartiailh, 2018-01-16:
Updating VISA does not interfere with Python, so you can do it safely. If you have time you could also try to update to the last release of Python 2.7 (2.7.14), even if I doubt it will help.

mandalav, 2018-08-10:
Hi,

After updating pyvisa, below is the visa info.

Machine Details:
   Platform ID:    Windows-XP-5.1.2600-SP3
   Processor:      x86 Family 6 Model 23 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.11
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          Dec  5 2015 20:32:19 (#v2.7.11:6d1b6a68f775)
   Unicode:        UCS2

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5242883
         Spec. Version: 4194304

I am still getting access violation error 0x000005F8.

Please let me know how to proceed further.

Thanks,
Venkat

MatthieuDartiailh, 2018-08-13:
Hi @mandalav,

Sadly I have been unable to make any progress on this. In particular, I do not have access to a windows 32 bits machine. From my previous investigations, I do not remember seeing any wrong with the types used in the ctypes binding. But I still believe that this is where we are doing something wrong. So the first place to look at is the visa header (visa.h can be found online) and check that we use the proper types for everything (in particular the type used to store the session id I would say).
I will try to dive again into this but if you have time to investigate on your side it may be more efficient since you have access to a test machine.

MatthieuDartiailh, 2020-09-06:
Since the last post I have seen report of users using Windows 32 bits successfully so it may have been a ctypes issue or a bad installation of either Python or VISA. Closing but feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh