pyvisa/pyvisa #272: VI_ERROR_SYSTEM_ERROR Cannot run pyvisa

gvoigt, 2017-07-18:
I installed ni-visa (via aur packages on arch linux) and pyvisa via pip. 
When I run `python -m visa info` I get: 
```
Machine Details:
   Platform ID:    Linux-4.11.9-1-ARCH-x86_64-with-arch
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.6.1
   Compiler:       GCC 6.3.1 20170306
   Bits:           64bit
   Build:          Mar 27 2017 00:27:06 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/vxipnp/linux/lib64/libvisa.so:
         found by: user
         bitness: 64
         Could not get more info:
            VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
```
Unknown system error doesn't tell me anything. What can I do about the error?

ritman, 2018-03-07:
Same thing happens for me.
~~~~
$ python -m visa info
Machine Details:
   Platform ID:    Linux-4.15.7-1-ARCH-x86_64-with-arch
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.6.4
   Compiler:       GCC 7.2.1 20171224
   Bits:           64bit
   Build:          Jan  5 2018 02:35:40 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /opt/ni-visa/usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
~~~~

MatthieuDartiailh, 2018-03-07:
Honestly I have no idea why. I will try to set up and VM and have a look (hopefully before next week). In the meantime, could you add the  following to https://github.com/pyvisa/pyvisa/blob/master/pyvisa/ctwrapper/highlevel.py#L115 ?
```python
from traceback import print_exc
print_exc()
```
and post again the output of `python -m visa info`

ritman, 2018-03-07:
Here it is. Let me know if you need anything else!

~~~~
$ python -m visa info
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/pyvisa/ctwrapper/highlevel.py", line 109, in get_debug_info
    sess, _ = lib.open_default_resource_manager()
  File "/usr/lib/python3.6/site-packages/pyvisa/ctwrapper/functions.py", line 1227, in open_default_resource_manager
    ret = library.viOpenDefaultRM(byref(session))
  File "/usr/lib/python3.6/site-packages/pyvisa/ctwrapper/highlevel.py", line 190, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
Machine Details:
   Platform ID:    Linux-4.15.7-1-ARCH-x86_64-with-arch
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.6.4
   Compiler:       GCC 7.2.1 20171224
   Bits:           64bit
   Build:          Jan  5 2018 02:35:40 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.0.dev0

Backends:
   ni:
      Version: 1.10.0.dev0 (bundled with PyVISA)
      #1: /opt/ni-visa/usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
~~~~

MatthieuDartiailh, 2018-03-07:
Out of curiosity does the AUR package also install NI-MAX ? If it does can you try starting it and using it to talk to an instrument ? This would be a way to check that everything is fine with the VISA installation.

ritman, 2018-03-08:
The AUR package unfortunately doesn't include NI-MAX.

MatthieuDartiailh, 2018-03-21:
I finally managed to setup a VM with Manjaro (an ArchLinux variant) and installed the ni-visa package. Right after installing I got the same error you have. However after restarting the VM the output of `python -m visa info` is now the following and everything looks fine. This was on fresh install after doing all updates. Did you restart after installing ?

```
Machine Details:
   Platform ID:    Linux-4.14.27-1-MANJARO-x86_64-with-arch-Manjaro-Linux
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/marul/anaconda3/envs/tests/bin/python
   Version:        3.6.4
   Compiler:       GCC 7.2.0
   Bits:           64bit
   Build:          Mar 13 2018 01:15:57 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.0.dev0

Backends:
   ni:
      Version: 1.10.0.dev0 (bundled with PyVISA)
      #1: /opt/ni-visa/usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
```

ritman, 2018-03-22:
The same actually work for me too after restarting. Sorry for not having tested this earlier. I guess this can be considered not to be a bug. Sorry for the troubles and thank for testing!

MatthieuDartiailh, 2018-03-22:
Yeah no problem. I learnt some stuff about arch in the the process. Could you perharps add a note in the docs (around installing) about using aur package on Arch based distros.

ritman, 2018-03-22:
I added a comment on AUR and asked the package maintainer to add a comment.

MatthieuDartiailh, 2018-03-22:
Thanks. I will add a note in PyVISA documentation also.

t-onoz, 2018-03-22:
Hi, I am maintaining ni-visa package in AUR.
As discussed here, logout & login is necessary after installing ni-visa in AUR.
This is because one needs to set an environment variable (`export VXIPNPPATH=/opt/ni-visa/usr/local/vxipnp`) that is required when installing NI-VISA under a non-standard (different from /usr/local/vxipnp) directory.

Actually, I have already included comments in the post-install script that says:

> ===> 
>===> Please re-login to renew environment variable.
>===> If USB-based devices do not run properly, check the following:
>===> # gpasswd -a <user> usbtmc
>===> # rmmod usbtmc
>===> refs: http://digital.ni.com/public.nsf/allkb/A10090850FF0887A86257926007A7E96
>===>       https://github.com/python-ivi/python-usbtmc#configuring-udev
>===> 

But this might not have been enough as it caused some confusion among users.
I will add a comment on AUR page as well.

Also, please be noted that the package on AUR installs just a small portion of NI-VISA packages (nivisa and nivisa-32bit rpms) so some functionalities may not work.
For a-bit-more-featured packages, please see [my repository](https://github.com/t-onoz/ni-visa.PKGBUILD).

Alternatively, if you want a full installation of NI-VISA 15.5, check out [this thread in NI forum](https://forums.ni.com/t5/Linux-Users/Using-NI-VISA-with-Arch-Linux-or-Ubuntu-14-04/gpm-p/3462358), though I think that older kernels (`linux-lts44` and `linux-lts44-headers` in AUR) are necessary to properly build kernel modules.

MatthieuDartiailh, 2018-03-22:
Out of curiosity, what interfaces are supported ? I read some mentions about PCI/PXI not being supported but are the most commonly used serial, GPIB, USB, and TCPIP all supported ? Thanks for clarifying things on the AUR package, I will take care of updating PyVISA documentation.

t-onoz, 2018-03-22:
I'm sorry but I just tested USBTMC because that was all I needed (I own GPIB-USB-HS but it is not supported with linux > 2.6.24).
I would appreciate it if  someone who uses other protocols tests it and gives some feedback.

If you are going to add notes on PyVISA documentation, please keep in mind that this package on AUR is still very premature 😞

MatthieuDartiailh, 2018-03-22:
Oh thanks for letting me know. I will try to test the TCPIP interface and let you know. And I will mention the fact it is not yet mature.

MatthieuDartiailh, 2018-03-28:
Ok I tested the TCPIP communication with the AUR package and I can confirm it works. See #323 for the documentation update.

closed 2018-03-28 by MatthieuDartiailh

greyltc, 2018-04-04:
Hey @t-onoz I was the previous maintainer of the ni-visa package in the AUR. I'm using your updated package now with a GPIB-USB-HS adapter, but somehow the adapter is not recognized by `rm.list_resources()`. What does your `lsusb -v` show for this device?
mine is:
```
# lsusb -v -s 003:011

Bus 003 Device 011: ID 3923:709b National Instruments Corp. GPIB-USB-HS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x3923 National Instruments Corp.
  idProduct          0x709b GPIB-USB-HS
  bcdDevice            1.01
  iManufacturer           1 National Instruments
  iProduct                2 GPIB-USB-HS
  iSerial                 3 018BA835
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           53
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

t-onoz, 2018-04-04:
@greyltc
Unfortunately, GPIB-USB-HS is only supported in very old (< 2.6.24) linux kernels, so you need to use alternative drivers like linux-gpib instead of NI-VISA.

https://forums.ni.com/t5/Linux-Users/NI-VISA-and-linux-gpib-requirements-technical-and-legalese/gpm-p/3375826