pyvisa/pyvisa #275: Assigning parameters for write_ascii_values in pyvisa for serial port

RolandSc, 2017-08-09:
Hi all,
 
I want to control my Keithley 6485 Picoammeter externally by connecting it via RS232toUSB to my Linux PC (CentOS 6.9) and writing a code in python (version 2.7.13) with pyvisa:
 
> import sys
> import visa
> from visa import constants
> 
> rm = visa.ResourceManager('/usr/local/vxipnp/linux/lib64/libvisa.so')
> #open serial connection and set baud to 9600, 8 data bits, CR termination, one stop bit, none parity, no flow control
> amm = rm.open_resource('ASRL2::INSTR', baud_rate = 9600, data_bits = 8, write_termination= '\r', read_termination = '\r')
> constants.VI_ASRL_STOP_ONE     
> constants.VI_ASRL_PAR_NONE
> constants.VI_ASRL_FLOW_NONE                  
> amm.write("*RST")                       # Return 6485 to RST default
> amm.write("TRIG:DEL 0")                 # Set trigger delay to zero seconds
> amm.write("TRIG:COUNT 2500")            # Set trigger count to 2500
> amm.write("SENS:CURR:RANG:AUTO OFF")    # Turn auto range off
> amm.write("SENS:CURR:NPLC .01")         # Set integration rate to NPLC 0.01
> amm.write("SENS:CURR:RANG 2e-7")        # Use 200 nA range 
> amm.write("SYST:ZCH OFF")               # Turn zero check off
> amm.write("SYST:AZER:STAT OFF")         # Turn auto zero off
> amm.write("DISP:ENAB OFF")              # Turn Display off
> amm.write("*CLS")                       # Clear status model
> amm.write("TRAC:POIN 2500")             # Set buffer size to 2500
> amm.write("TRAC:CLE")                   # Clear buffer
> amm.write("TRAC:FEED:CONT NEXT")        # Set storage control to start on next reading
> amm.write("STAT:MEAS:ENAB 512")         # Enable buffer full measurement event
> amm.write("*SRE 1")                     # Enable SRQ on buffer full measurement event 
> amm.write("INIT")                       # start taking and storing readings wait for GPIB SRQ line to go true
> amm.write("DISP:ENAB ON")               # Turn display on
> print(amm.query_ascii_values("TRAC:DATA?")) # Request data from buffer

When I execute my script I get the following error:

> File "./AMDAQ.py", line 31, in <module> print(amm.query_ascii_values("TRAC:DATA?"))
> pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

I know that the device is just communicating with ASCII and the readout should look like this: Reading, Timestamp, Status and the error message after amm.write("*RST"): -113 undefined header. So I think the messages and not transferred correctly.
I know over the RS-232 interface, only the ASCII format is allowed. But when I follow the example in the pyvisa instruction with write_ascii_values(text, values) and assigning it a list, I only get an error message from the device -100 Command error.
Can somebody please tell me how to set the variables in write_ascii_values correctly or what I am doing wrong? Are my settings for the serial device wrong? Sometimes when I execute 2 times I get the error "VI_ERROR_ASRL_FRAMING (-1073807253): A framing error occurred during transfer ." too. I just do not know what to do.
Thank you!
Regards, Roland

MatthieuDartiailh, 2017-08-09:
Can you read anything from your device ? Such as let's say the state of the display (I guess amn.query('DISP:ENAB?') should give you that info).
If you also get a Timeout erro on this command it may mean that the read_termination is wrong. Also note write_ascii_values is meant to sent list of values to an instrument not to read them as you are trying to do in amm.query_ascii_values("TRAC:DATA?").

RolandSc, 2017-08-09:
Hi Matthieu,

Thanks for your answer!
It depends: For example in a previous version of my code I had the line amm.query(":*OPC?") (operation complete query), before the INIT data taking. Then I got always a "1" as output when I do print (amm.query("TRAC:DATA?")). However, when I ask amm.query("*IDN?") I get the timeout error. So I guess, always when there is something more involved than just a number, it cannot read it. 

I suspected a termination issue, but my device is terminated with CR and I assigned the read_termination = '\r', so I do not know what I can change (I also tried it with \n, but it did not help).

When I type in write_ascii_values it says TypeError: write_ascii_values() takes at least 3 arguments (2 given). I do not know what I should assign as the third argument (numpy.array?).

Regards, Roland

MatthieuDartiailh, 2017-08-09:
I think the first thing to do is fixing the termination issue. You can try to read the instrument answer one byte at a time using the following:
```python
amm.write("*IDN?")
while True:
    print(amm.visalib.read(amm.session, 1))
```
If you do not get any answer it may be that the write_termination is wrong and that the instrument is waiting for something else.

RolandSc, 2017-08-09:
Hi Matthieu,

thanks again! So for the following: I set the device termination to CR ('\r'). If I do not assign any termination in the script I get: 

> /usr/local/lib/python2.7/site-packages/pyvisa/ctwrapper/functions.py:1582: VisaIOWarning:
> VI_SUCCESS_MAX_CNT (1073676294): The number of bytes transferred is equal to the requested input count. More data may be available.
> ret = library.viRead(session, buffer, count, byref(return_count))
> ('K', <StatusCode.success_max_count_read: 1073676294>)
> ('E', <StatusCode.success_max_count_read: 1073676294>)
> ('I', <StatusCode.success_max_count_read: 1073676294>)
> ('T', <StatusCode.success_max_count_read: 1073676294>)
> ('H', <StatusCode.success_max_count_read: 1073676294>)
> ('L', <StatusCode.success_max_count_read: 1073676294>)
> ('E', <StatusCode.success_max_count_read: 1073676294>)
> ('Y', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
> ('I', <StatusCode.success_max_count_read: 1073676294>)
> ('N', <StatusCode.success_max_count_read: 1073676294>)
> ('S', <StatusCode.success_max_count_read: 1073676294>)
> ('T', <StatusCode.success_max_count_read: 1073676294>)
> ('R', <StatusCode.success_max_count_read: 1073676294>)
> ('U', <StatusCode.success_max_count_read: 1073676294>)
> ('M', <StatusCode.success_max_count_read: 1073676294>)
> ('E', <StatusCode.success_max_count_read: 1073676294>)
> ('N', <StatusCode.success_max_count_read: 1073676294>)
> ('T', <StatusCode.success_max_count_read: 1073676294>)
> ('S', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
> ('I', <StatusCode.success_max_count_read: 1073676294>)
> ('N', <StatusCode.success_max_count_read: 1073676294>)
> ('C', <StatusCode.success_max_count_read: 1073676294>)
> ('.', <StatusCode.success_max_count_read: 1073676294>)
> (',', <StatusCode.success_max_count_read: 1073676294>)
> ('M', <StatusCode.success_max_count_read: 1073676294>)
> ('O', <StatusCode.success_max_count_read: 1073676294>)
> ('D', <StatusCode.success_max_count_read: 1073676294>)
> ('E', <StatusCode.success_max_count_read: 1073676294>)
> ('L', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
> ('6', <StatusCode.success_max_count_read: 1073676294>)
> ('4', <StatusCode.success_max_count_read: 1073676294>)
> ('8', <StatusCode.success_max_count_read: 1073676294>)
> ('5', <StatusCode.success_max_count_read: 1073676294>)
> (',', <StatusCode.success_max_count_read: 1073676294>)
> ('4', <StatusCode.success_max_count_read: 1073676294>)
> ('3', <StatusCode.success_max_count_read: 1073676294>)
> ('1', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('5', <StatusCode.success_max_count_read: 1073676294>)
> ('6', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> (',', <StatusCode.success_max_count_read: 1073676294>)
> ('C', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('1', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
> ('J', <StatusCode.success_max_count_read: 1073676294>)
> ('u', <StatusCode.success_max_count_read: 1073676294>)
> ('n', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
> ('2', <StatusCode.success_max_count_read: 1073676294>)
> ('3', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
> ('2', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('1', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> (' ', <StatusCode.success_max_count_read: 1073676294>)
[... 16 lines cut ...]
> ('\r', <StatusCode.success_max_count_read: 1073676294>)
> Traceback (most recent call last):
>   File "./AMDAQtest.py", line 17, in <module>
>     print(amm.visalib.read(amm.session, 1))
>   File "/usr/local/lib/python2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1582, in read
>     ret = library.viRead(session, buffer, count, byref(return_count))
>   File "/usr/local/lib/python2.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
>     raise errors.VisaIOError(ret_value)
> pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

With the read_termination='\r', the last line before the error changes to

> ('\r', <StatusCode.success: 0>)
> Traceback (most recent call last): File "./AMDAQtest.py", line 17, in <module>
print(amm.visalib.read(amm.session, 1))
File "/usr/local/lib/python2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1582, in read
ret = library.viRead(session, buffer, count, byref(return_count))
File "/usr/local/lib/python2.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation
completed.

Although all the information has been passed from the device and it says status code success with the matching termination, it still gives out the error. I do not know what this means. By the way, the write termination does not seem to matter, whatever I do. 

Regards,
Roland

MatthieuDartiailh, 2017-08-09:
That's really weird. If you directly query the IDN (amm.query('*IDN?')), do you get the timeout ? I am a bit lost as the termination character appear to be there so read should complete without issue on the *IDN? request.

RolandSc, 2017-08-09:
Yes, when I only do the query the error and timeout is still there. It is really strange. Are there more tools for debugging?

MatthieuDartiailh, 2017-08-09:
What version are you using ? Coiud you try installing from github ?
pip install https://github.com/pyvisa/pyvisa/tarball/master

RolandSc, 2017-08-10:
Ok, I installed it and now when I do the amm.query('*IDN?') it actually works and I get the ID of the picoammeter. However, when I tried it with my original code, the timeout error appears again. Also when I use
`amm.write("*IDN?")
while True:
    print(amm.visalib.read(amm.session, 1))`
I get the timeout error.

So is there something wrong with the processing of data at the very end, or is it because I am using python 2.7?

Regard,
Roland

MatthieuDartiailh, 2017-08-10:
Can you post the version of the script that works and the one that fail ? I am a bit lost in what you tried. Also I noticed in your original script you do:
```python
    amm = rm.open_resource('ASRL2::INSTR', baud_rate = 9600, data_bits = 8, write_termination= '\r', read_termination = '\r')
    constants.VI_ASRL_STOP_ONE
    constants.VI_ASRL_PAR_NONE
    constants.VI_ASRL_FLOW_NONE
```
Accessing the constants attribute does not set anything on your instrument, so maybe something is wrong here.

RolandSc, 2017-08-10:
I played around a little bit more, but step by step:

`amm.query("*IDN?")`
gives me the output:
KEITHLEY INSTRUMENTS INC.,MODEL 6485,4310560,C01   Jun 23 2010 12:22:00/A02  /J

When doing this:
`amm.write("*IDN?")`
`while True: print(amm.visalib.read(amm.session, 1))`
I get the output I posted in my pre last answer: Time out error after the line 
`('\r', <StatusCode.success_max_count_read: 1073676294>)`

But now when I use the original code (withouth the constants attributes) with
`print(amm.query(":TRAC:DATA?"))`  and I set the trigger and buffer count to 2 I get this:
`+4.115464E-11A,+0.000000E+00,+0.000000E+00,+6.184564E-11A,+0.000000E+00,+0.000000E+00`
So this is the right output I want to have.
I raised the number till 20 and it is fine, but for 30 I get the time out error, so it has some problems with sending a bigger amount of data...

MatthieuDartiailh, 2017-08-10:
What happens if you read byte by byte to get the Trac data after how many does it fail ?

RolandSc, 2017-08-10:
Already after I set the Count to 1:

> ('+', <StatusCode.success_max_count_read: 1073676294>)
> ('2', <StatusCode.success_max_count_read: 1073676294>)
> ('.', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('2', <StatusCode.success_max_count_read: 1073676294>)
> ('3', <StatusCode.success_max_count_read: 1073676294>)
> ('6', <StatusCode.success_max_count_read: 1073676294>)
> ('2', <StatusCode.success_max_count_read: 1073676294>)
> ('6', <StatusCode.success_max_count_read: 1073676294>)
> ('E', <StatusCode.success_max_count_read: 1073676294>)
> ('-', <StatusCode.success_max_count_read: 1073676294>)
> ('1', <StatusCode.success_max_count_read: 1073676294>)
> ('1', <StatusCode.success_max_count_read: 1073676294>)
> ('A', <StatusCode.success_max_count_read: 1073676294>)
> (',', <StatusCode.success_max_count_read: 1073676294>)
> ('+', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('.', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('E', <StatusCode.success_max_count_read: 1073676294>)
> ('+', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> (',', <StatusCode.success_max_count_read: 1073676294>)
> ('+', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('.', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('E', <StatusCode.success_max_count_read: 1073676294>)
> ('+', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('0', <StatusCode.success_max_count_read: 1073676294>)
> ('\r', <StatusCode.success: 0>)
> Traceback (most recent call last):
>   File "./AMDAQ.py", line 37, in <module>
>     print(amm.visalib.read(amm.session, 1))
>   File "/usr/local/lib/python2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1582, in read
>     ret = library.viRead(session, buffer, count, byref(return_count))
>   File "/usr/local/lib/python2.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
>     raise errors.VisaIOError(ret_value)
> pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2017-08-10:
The last timeout is expected as we have seen the termination but kept reading. I am curious of what happens of if you set the count to 30 ? When does the timeout occur in such a case ?

RolandSc, 2017-08-10:
So I raised the count to 30, 100 and 1000 and each time the error occurs after the termination line;
`('\r', <StatusCode.success: 0>)`

MatthieuDartiailh, 2017-08-10:
This is truly weird ! I will try to give a look as the code in charge of the reading but I start to wonder if the instrument does not wait for some data to be read to send the following. You could try to increase the timeout (to something like 100 ms) and see if it works. Otherwise do the reading in a manual fashion and stop at the termination character.

RolandSc, 2017-08-10:
Hi Matthieu,

The increase of the timeout was the way to go. Although I set it to 100.000, it took ages for the 1000 counts but it worked! So I guess I just play around with this number to see when it full fills my needs.

Thank you very much for help! I really appreciate it! Have a good day, I see this issue now as resolved!

Regards,
Roland

closed 2017-08-10 by RolandSc