pyvisa/pyvisa #276: How do you make sure you don't open a resource twice?

geib7592, 2017-08-09:
I'm having trouble figuring out to avoid opening a resource that was open before and not yet closed. I think my question is related to [this unanswered StackOverflow question](https://stackoverflow.com/questions/41501356/pyvisa-isopen-function).

Basically, I've written a function that opens a particular instrument like this:
```python
rm = visa.ResourceManager()
instr = rm.open_resource(address)
```
Ideally, I would modify the above so that it first checks to see if a connection is already made to instrument, and if so, use the existing connection. Is there a straightforward way to implement this? 

In the linked StackOverflow question, the poster mentions getting an exception when trying to open the resource, but I'm finding that I'm able to open the resource multiple times, and I only get errors when I start trying to send commands to the instruments.

MatthieuDartiailh, 2017-08-10:
I do not think it is possible. You have to keep track of the resources yourself as neither the resource manager nor the library keep track of those at the Python level

geib7592, 2017-08-10:
I see. Thanks for the info!

closed 2017-08-10 by geib7592