pyvisa/pyvisa #277: 64 bit memory read on PCIe

monticello0427, 2017-09-18:
I'm having difficulty reading a 64 bit register with PyVISA. I've had no issues reading other registers when I set the length to 32-bit and lower. However, when I try 64-bit, pyvisa raises a "NotImplementedError" exception. My PC is running 64-bit and the bitness of python and the NI-VISA library are both 64-bit, so I don't believe it is a bit mismatch error. 

Here is some sample code:

```
rm = visa.ResourceManager()
board = rm.open_resource(rm.list_resources()[0])

board.write_memory(11,int(0x0000),int(0xdeadbeef),32)
board.write_memory(11,int(0x0004),int(0xaa55aa55),32)
rf_card.read_memory(11,int(0x0000),64)[0]

```

I'm running pyvisa 1.8 and python 3.6 to execute the code.

Thank you in advance for any help you can offer!

MatthieuDartiailh, 2017-09-18:
The needed function appear to be missing from the list of functions visa_functions found in https://github.com/pyvisa/pyvisa/blob/master/pyvisa/ctwrapper/functions.py. Along with all the other 64 bits functions (move_in, move_out, peek, etc).
Could you please submit a PR with the appropriate changes ? In the meantime @hgrecco any reasons why those functions were not included ?

monticello0427, 2017-09-18:
Thank you for the quick reply. I'll submit that as soon as possible.

monticello0427, 2017-09-18:
That worked. I'll submit the change in a bit. Thanks, @MatthieuDartiailh. However I have noticed there is no functionality for writing 64-bit values, only for reading them. Is this a limitation or something I can also add?

monticello0427, 2017-09-18:
Thanks @MatthieuDartiailh . I am now able to read 64 bit values as well as write them. I'll submit the PR later today.

MatthieuDartiailh, 2017-11-05:
Closed by #278

closed 2017-11-05 by MatthieuDartiailh