pyvisa/pyvisa #280: Regular expression syntax is not documented

awneil, 2017-10-04:
The Resource Manager documentation for list_resources() and list_resources_info() states that the 'query' parameter is a regular expression - but gives no definition or description or reference for the syntax of that regular expression

http://pyvisa.readthedocs.io/en/stable/api/resourcemanager.html#pyvisa.highlevel.ResourceManager.list_resources

I did eventually stumble upon this: https://github.com/pyvisa/pyvisa/blob/master/pyvisa/rname.py - which does contain a definition of the regular expression syntax.

It also points out that the regular expression syntax used here is **_not_** the same as the standard Python regular expression syntax - so it is all the more important that this should be stated in the documentation!

awneil, 2017-10-04:
The VISA Resource Regular Expression is defined (with examples) in **VPP-4.3: The VISA Library specification** 
Section 4.4.1
Tables 4.4.1 - 4.4.4

http://www.ivifoundation.org/specifications/

See also:  http://zone.ni.com/reference/en-XX/help/370131S-01/ni-visa/findingvisaresourcesusingregularexpressions/

And: http://zone.ni.com/reference/en-XX/help/370131S-01/ni-visa/visaresourcesyntaxandexamples/

MatthieuDartiailh, 2017-10-05:
Would you mind doing a pull request to add this to the docs ?

awneil, 2017-10-05:
I wouldn't mind, but someone would have to walk me through that process - I don't do Git

MatthieuDartiailh, 2017-10-05:
First you need to fork the repo (Fork button on the Github UI), then clone locally the forked repository (that you own). Create a new branch (regular-docs could be a good name but it does not really matter). Update the documentation files that need to be updated. Then commit your changes and push them to your repo. Once this is done the Github UI on your repo will ask you if you want to open a PR, just click on the button. If you are looking for git graphical interface, I would recommend Smartgit.

awneil, 2017-10-05:
OK ...

MatthieuDartiailh, 2017-10-05:
If you have more questions do not hesitate. Actually another simpler option would be to fork the project then simply edit the docs directly on Github, once you are done Github will handle the commit and branch creation for you and you will just have to create the PR.
If you can't do it I will probably do it at some point but I cannot say when.

awneil, 2017-10-05:
Is it sufficient just to update the docstrings for list_resources() and list_resources_info() in pyvisa/highlevel.py ?

MatthieuDartiailh, 2017-10-05:
I think that it is sufficient. The fact that one can filter the resources is not mentioned anywhere in the docs, which should also probably be fixed by adding a note in the tutorial. If you are familiar with sphinx (the tool used to build the documentation) you can add one. Otherwise I will add it to your PR myself.

MatthieuDartiailh, 2017-11-05:
@awneil are you still planning to work on this ?

awneil, 2017-11-06:
oops!  thanks for the riminder - yes!

MatthieuDartiailh, 2017-11-21:
ping @awneil

MatthieuDartiailh, 2017-11-27:
ping @awneil

MatthieuDartiailh, 2018-01-25:
ping @awneil

MatthieuDartiailh, 2018-02-25:
ping @awneil

awneil, 2018-03-15:
I hope that's it ... ?

MatthieuDartiailh, 2018-03-28:
Closed by #320

closed 2018-03-28 by MatthieuDartiailh