pyvisa/pyvisa #287: pyvisa module not found

mrasyrafafandi, 2017-11-03:
where can i find pyvisa module? I have searched everywhere in my documents and the only visa.py is available.
this is the result when i run the command

 ` runfile('C:/Users/user/Downloads/untitled01.py', wdir='C:/Users/user/Downloads')
Traceback (most recent call last):

  File "<ipython-input-1-cb3ada0c104d>", line 1, in <module>
    runfile('C:/Users/user/Downloads/untitled01.py', wdir='C:/Users/user/Downloads')

  File "C:\Users\user\Anaconda3\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 699, in runfile
    execfile(filename, namespace)

  File "C:\Users\user\Anaconda3\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 88, in execfile
    exec(compile(open(filename, 'rb').read(), filename, 'exec'), namespace)

  File "C:/Users/user/Downloads/untitled01.py", line 22, in <module>
    import visa, time, datetime, os, winsound

  File "C:\Users\user\Anaconda3\lib\visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants

ImportError: No module named 'pyvisa'`

MatthieuDartiailh, 2017-11-13:
How did you install Pyvisa (and which version) ?

MatthieuDartiailh, 2018-02-25:
Closing feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh