pyvisa/pyvisa #288: No example for open_raw_resource

lucask07, 2017-11-05:
I'm not able to open an older Agilent instrument over USB. `open_resource` fails, `open_raw_resource` works, but there is no example for how to use `open_raw_resource`.
Could future documentation include an example for how to use `open_raw_resource`?

I've posted my attempts on stackoverflow:
https://stackoverflow.com/questions/47105179/pyvisa-cannot-open-agilent-n9320b-over-usb

MatthieuDartiailh, 2017-11-05:
By default PyVisa only list instruments supporting the INSTR protocol for USB (and in this match the behavior of the visa library). However you can pass the resource you get from ```list_resources('?*')``` directly to ```rm.open_resource``` just like any other resource.

```open_raw_resource``` is meant to only give a visa session to use at a somehow lower level and is not what you are after here.

lucask07, 2017-11-05:
`open_resource(usb_id)` (using the instrument ID derived from the Mac System Report ```usb_id = 'USB0::0x0957::0xffEF::CN03224880::INSTR' ```) fails with:
```
/Applications/anaconda/lib/python2.7/site-packages/pyvisa/highlevel.pyc in open_resource(self, resource_name, access_mode, open_timeout, resource_pyclass, **kwargs)
   1642                 raise ValueError('%r is not a valid attribute for type %s' % (key, res.__class__.__name__))
   1643 
-> 1644         res.open(access_mode, open_timeout)
...
VisaIOError: VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.
```
Tomorrow, I will retry `open_resource` with the instrument ID returned by `list_resources('?*')` which is different from the System Report address. Why are these addresses different?
```
In [3]: rm.list_resources('?*')
Out[3]: 
(u'USB0::0x05AC::0x8600::NI-VISA-337641472::0::RAW',
 u'USB0::0x05AC::0x8600::NI-VISA-337641472::1::RAW',
```
Thank you

MatthieuDartiailh, 2017-11-13:
I have honestly no idea why the addresses are different. Did you get the system to work ?

lucask07, 2017-11-13:
Hi Matthieu,
No, not yet. Opening the address returned with the NI backend and ‘?*’ did
not fail. However the instrument did not respond to commands sent.
My next step is to try with Agilent VISA libraries...but that requires a
Windows machine.

Best,
Lucas

MatthieuDartiailh, 2018-01-25:
Any update on this ?

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh