pyvisa/pyvisa #291: pyvisa.errors.LibraryError: Error while accessing visa32.dll: [Error 126]

h8d9123, 2017-11-13:
**code:**
import visa
rm = visa.ResourceManager("visa32.dll")

**- error:**
File "<string>", line 1, in <module>
File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 226, in __new__
  visa_library = VisaLibrary(visa_library)
File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 102, in __new__
  raise errors.LibraryError.from_exception(exc, library_path)
visa.errors.LibraryError: Error while accessing visa32.dll: [Error 126]

 I tried pyvisa1.5 and 1.8, but the results are same.

MatthieuDartiailh, 2017-11-13:
Not sure what is happening here. Why do you specify the visa implementation to use ? Usually Pyvisa does the right thing an pick up the proper dll. Also did you post the full error message ? it looks truncated to me.

h8d9123, 2017-11-13:
thank you,MatthieuDartiailh .
this is all info:
>>> import visa
>>> rm = visa("visa32.dll")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'module' object is not callable
>>> rm = visa.ResourceManager("visa32.dll")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 226, in __new__

    visa_library = VisaLibrary(visa_library)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 102, in __new__

    raise errors.LibraryError.from_exception(exc, library_path)
pyvisa.errors.LibraryError: Error while accessing visa32.dll: [Error 126]

**Question:**
the software ,ni-visa, is necessary?
pyvisa can be use when ni-visa is installed. but I just want to use some about gpib.

MatthieuDartiailh, 2017-11-13:
To communicate through GPIB on windows you need to install the VISA library (that pyvisa simply calls). You do not need the full National instrument stack but you do need ni-visa (or another implementation of visa).

h8d9123, 2017-11-13:
Thank you very much, MatthieuDartiailh

closed 2017-11-13 by h8d9123